hhd-dev/hhd #2: Swap_legion appears to function opposite as described.

th3raid0r, 2023-12-26:
As in description, swap_legion isn't working correctly for me. It also is using terms that are unique to DualShock 4 and older controllers (select and start don't exist anymore, those buttons are "create" and "options" now). Unless, of course this is meant to simulate clicking on the left or right of a DS5 trackpad - despite the steam UI suggesting the "create" and "options" buttons being activtated.

I'll assume that "create" == "select" and "options" == "start" since that appears to be what you mean. 

In that case, things still don't line up:

- **True / "l_is_select":** Top-Left button (legion L) is QAM, Top-Right button is "Guide". Select/Create is mapped to the hamburger menu button on the left. - I'd instead expect that the Top Left button is "select". 
- **"l_is_start":** Top-Left button (legion L) is QAM, Top-Right button is "Guide". Select/Create is mapped to the hamburger menu button on the left. - I'd instead expect that the Top Left button is "start"/"options". 
- **False**: Top-Left button (legion L) is Create/Select. Top-Right is Options/Start. - I'd instead expect this result for "True or l_is_select".

I suspect that the issue could be related to the NoneType you used here - https://github.com/antheas/hhd/blame/3a0e7d2607c97d49d4f51c1a25809d1d985a723f/src/hhd/device/legion_go/base.py#L309

But I haven't had a chance to confirm that yet by trying it locally.

th3raid0r, 2023-12-26:
Oh, and I'm on v0.1.11 - I see v0.2.0 is on the way, but I don't think it will change the issue.

th3raid0r, 2023-12-26:
NVM - it looks like this may be fixed - dunno how I missed it (I check each commit so I know how this darn thing works!)

```
    match conf["swap_legion"].to(str):
    ...
```

That area has much better parsing now. Going to upgrade and report back.

antheas, 2023-12-26:
Hi Brian,
you can upgrade now and test it out, otherwise it's gonna hit tomorrow. I have not changed anything though in terms of that. I just tested it and it works the way I expect.

If you used the local install it works like this:
```bash
~/.local/share/hhd/venv/bin/pip install git+https://github.com/antheas/hhd
```

But that way may not be the way you expect. Since v0.2 will break everyone's configs this is the time to fix any ambiguous names. So I would love your feedback.

A bit about the naming:
HHD is designed to be general use and to support more controllers in the future. So I am trying to use generic names for all the gamepad buttons.

The names I came up with (by referencing kernel + controllers) are:
 - guide: to be the primary vendor button of the controller (e.g., playstation, xbox, steam button)
 - share: to be the secondary vendor button of the controller (QAM, share, mute)
 - start: to be the right controller button that typically pauses the game (start, options)
 - select: to be the left controller button that does a secondary game action (e.g., open a map)

The option swap legion does the most requested community option for the Legion go, which was also added natively in windows in the latest update.

It swaps the top buttons of the controller, with the bottom left buttons. Meaning, it swaps Guide with Start and Share with Select. This is what True does. And this is why the option is named `guide_to_start` internally.

However, since the primary button of the go is on the left, this does not feel natural. Because the left button becomes start and the right select, they are conventionally reversed. Which is where the two extra options come in:
- (legion) `l_is_start`: remaps Legion L to start and R to select
- (legion) `l_is_select`: remaps Legion R to start and L to select (conventional)

I tested the last v0.2 version and verified that this is what happens.

Funnily, the legion space version did the same oopse but in reverse. With the latest legion space update, the buttons work like `l_is_select` but Legion R is on the top and Legion L is on the bottom. Which is quite confusing.

So what did you expect the option to do? I'm going to review what you wrote again but it is a bit ambiguous.

Also, this option breaks HHD and the only way we know to reverse it is by using the windows version of legion space right now. Resetting the controllers does not work.

th3raid0r, 2023-12-26:
Huh, I definitely don't get the behavior you are describing, but it sounds like my expectations are in line with yours, but I did set the aforementioned option in Legion Space on windows. 

To make things abundantly clear:

| Setting Value | Expected Behavior | Actual Behavior |
|--------|--------|--------|
| True | remaps Legion R to start and L to select (conventional) | Top-Left button (legion L) is QAM, Top-Right button is "Guide".  |
| "l_is_start" | remaps Legion L to start and R to select | Top-Left button (legion L) is QAM, Top-Right button is "Guide". |
| False | Legion L remains "guide", Legion R remains QAM | Top-Left button (legion L) is Create/Select. Top-Right is Options/Start. | 

You might want to mention the conflict with Legion space in your Readme (or perhaps you have recently - you're moving pretty fast). 

I'm going to reboot, toggle it, and check to see if that's what it was.

th3raid0r, 2023-12-26:
Well, butter my biscuit, that's what was causin' all the ruckus!

Looks like I'll need to toggle that when I use my windows partition. I generally use the Legion Space button swap to make EmuDecks Yuzu button combos more feasible (Start + Select + LB +RB is REALLY hard when start and select are in the bottom left of the controller).

Closing since this is definitely not an issue on HHD's side.

closed 2023-12-26 by th3raid0r

antheas, 2023-12-26:
Also since that option is sticky if you use it you can just set swap legion. To false

I have not verified what that option does exactly in Linux but from what you're describing it works with no changes.

A future version of hhd will allow using the legion space remappings as well

But for that we have to get a gui for hhd going. The next version of hhd includes a rest API for this reason