hhd-dev/hhd #5: Doc nit: "Steam Deck sleep"

appsforartists, 2023-12-29:
> Short press makes steam deck sleep

Since this isn't designed to be used on a physical Steam Deck, this wording is a bit ambiguous.  Perhaps "short press initiates [sleep/hibernate/hybrid sleep](https://wiki.archlinux.org/title/Power_management/Suspend_and_hibernate)" (picking whichever one it does).

antheas, 2023-12-30:
Hi Brenton, the power button is something that is managed by your desktop environment and works fine without HHD. 

What HHD does is that it detects when steam is running in steam deck mode and if it is, it disables the power button and waits for it to be pressed. Once it is pressed, it commands steam to execute its hibernate/long press events.

This allows steam to show you the wink animation, and if your game supports continuous saves, to upload the saves to the cloud. Then, steam itself makes the device sleep.

If steam is not running in big picture mode then hhd does nothing.

I will leave this issue open until I do the disambiguation in the readme.

antheas, 2023-12-30:
In steamos, there is a bash script that does this on behalf of steam, since steam does not have this kind of priviledge

Handygccs does something similar for the power button, and in fact the hhd code is based on that.

However, HHD is more selective and only grabs the power button when steam is open, whereas handygccs does this always. This allows you to remap the power button in your DE as you wish.

In addition, hhd hooks the power button long press, which displays a power menu in steam big picture, and it is the first daemon other than the steamos implementation to do so.

appsforartists, 2023-12-30:
There's a [PR coming](https://github.com/Jovian-Experiments/Jovian-NixOS/pull/257) for Jovian that also adds Legion Go power button support.  https://github.com/Jovian-Experiments/powerbuttond/commit/4b5baf9dc574514a7c0363bcf13d86d943c3d338

It works for short presses, but the Legion Go is weird about long presses: from my short investigation, it appears that instead of sending `KEY_POWER` on `event2`, it sends `KEY_LEFTMETA` or `-1` on `event1`.

Sounds like you're taking care of that.  ❤️

antheas, 2023-12-30:
yeah, i spoke to the chimeraos developers about this.

It's not that weird actually, they told me it is very common for laptops/handhelds to do this. But nobody tries to remap the power button long press normally...

Its such a small feature it typically gets no love so it was not fixed until now.

antheas, 2023-12-30:
https://github.com/hhd-dev/hhd/commit/01543d651dc7c93455a0299587612991951aba06

closed 2023-12-30 by antheas