hhd-dev/hhd #6: Thoughts on adding xpad support to the kernel?

appsforartists, 2023-12-29:
I've submitted a patch to the kernel that [adds the Legion Controller USB IDs to `xpad`](https://github.com/appsforartists/linux/commit/e3731167d3caff0857e5700c9129e4d8a386aa6a).  This adds out-of-the-box support for all the buttons except {the rear paddles and the Legion buttons} to the Linux kernel.  Due to a quirk with my work email setup and the Linux kernel contribution process, [the patch](https://lore.kernel.org/lkml/CAAL3-=88exVfuL1Y-kvPNbsU+d-UTfDLFViWVObFLtbC4xueeA@mail.gmail.com/) is on hold until January.

I see 

> Hiding the original xbox controller!
 
in the README.  I presume this means `hhd` is compatible with the kernel patch, and I should carry-on submitting it when my email troubles are resolved next month.  Is this correct?

antheas, 2023-12-30:
Hi Brenton,
yes it is compatible with your patch and actually requires it to work. Right now, most distributions ship a udev rule that activates xpad manually because it does not have the PID/VID combinations you add. This fixes the xbox controller for now. But for example, Nobara 39 did not include this rule, which is why it is now part of the readme.

HHD actually reads the original controller using the xpad output, since the controllers are 500hz and it is not realistic to do this kind of processing outside the kernel and especially not in python.

Your patch is expected to change the title of the controllers, but I already prepared for this by switching to using a capabilities match instead of name.

antheas, 2023-12-30:
Good luck on getting the rule accepted!

closed 2023-12-30 by antheas