hhd-dev/hhd #10: Add Credits

appsforartists, 2023-12-30:
The `hid` module appears to be a fork of @apmorton's pyhidapi.

There should be a credits section in the README to cite prior work.

antheas, 2023-12-30:
`uhid` is also sourced from `python-uhid`. There is attribution on the license file. Both of the files have been essentially rewritten.

See license:
https://github.com/hhd-dev/hhd/blob/master/LICENSE

antheas, 2023-12-30:
I agree with you. We also need a contributor agreement and an expanded license section. Can you help with that?

The contributor agreement should list that contributors surrender their copyright, provided that it is always made available under an open source license variant, and that they do in fact have permission to contribute the content they do.

I do not particularly like that I chose MIT (I did it to not dual license and still include `hid`, `uhid`) and I am the sole copyright holder of hhd right now (other than `hid`, `uhid`). I would like to switch to a GPL variant in the future and the only way to do that is by creating a contributor agreement right now.

appsforartists, 2023-12-30:
I don't think a CLA is the best path forward here.

For people with corporate jobs, their employers like to claim copyright on any code they write (whether at work or extracurricularly).  Since it's open-source anyway, nobody goes through the effort to file an exemption (which would allow you to retain the copyright, but forbid you from using any corporate resources for future work).

For instance, at Google, we can patch [any project with an approved license](https://opensource.google/documentation/reference/patching) with no additional bureaucracy if there's no CLA.  If there is a CLA, you [have to contact the lawyers](https://opensource.google/documentation/reference/patching#cla) before submitting a patch.

appsforartists, 2023-12-30:
I don't think a CLA is the best path forward here.

For people with corporate jobs, their employers like to claim copyright on any code they write (whether at work or extracurricularly).  Since it's open-source anyway, nobody goes through the effort to file an exemption (which would allow you to retain the copyright, but forbid you from using any corporate resources for future work).

For instance, at Google, we can patch [any project with an approved license](https://opensource.google/documentation/reference/patching) with no additional bureaucracy if there's no CLA.  If there is a CLA, you [have to contact the lawyers](https://opensource.google/documentation/reference/patching#cla) before submitting a patch.

As I see it, there are three issues at play that are all exacerbated by this being a small project:

- The additional friction (going through a legal review before submitting) may be enough to prevent people from contributing.
- Lawyers probably aren't going to approve a hand-written CLA.  There's a long list of well-intentioned, self-written licenses (like the Do Whatever The Fuck You Like license) that are explicitly not approved because they're not legally correct.  That effectively bans contributors who have corporate jobs.
- The stakes are pretty small. MIT is the easiest license - nobody has to think too hard about it, and you've already forked MIT-licensed code for inclusion.  The target audience is a small group of individuals who are probably going to be using GitHub's forking system to share their modifications as PRs anyway.  I don't see a big risk of anyone making a bajillion dollars from Handheld Daemon and refusing the share their modifications.

As you've said, substantially all of the code in this repo is yours.  If you felt so inclined, you could re-release it under a different license, and either ask the other contributors to relicense theirs, or replace the forked code with e.g. imports and patches/wrappers.

Practically, I don't think any of this is worth it.  A CLA adds real friction that will cost you real contributors in search of ideological gains.  For a small project like this, not only is the loss of contributors a big cost, but there's not much to gain.  Whether it's MIT or GNU, most people are going to be sharing their code anyway.  The commercial/closed-source value of getting an esoteric gamepad to spoof some more mainstream ones doesn't seem like it's high enough to stress over this.

antheas, 2023-12-30:
Sure, all valid points. So I guess we stay with MIT. I am not that attached to GNU, other than noting the fact that all similar projects are under a GNU license. I did not know it is this big of a mess to be honest. Just wanted to cover myself legally.

I will leave this issue open to fix the readme.

antheas, 2024-10-13:
Fixed a couple days ago. MIT is compatible with GPL so that was no biggie, just needs the headers for that. I also added credits.

You were both right and wrong about the CLA. You were right in that it has a very bad smell. As for more contributors due to not having a CLA, I'd say it's still me so it did not have much of an effect.

closed 2024-10-13 by antheas