hhd-dev/hhd #8: Wrong buttons being dispatched

appsforartists, 2023-12-30:
I've just [packaged this for NixOS](https://github.com/NixOS/nixpkgs/pull/277661/) and tested it.

The buttons Steam sees don't match the physical location on the device.  For instance:

- The shape buttons are rotated: circle (back) is the physical Y button.
- The Legion buttons are registering as L3 (stick click and chorded stick click).
- The triggers are registering as the right stick.
- The right stick is registering as the triggers.

I tried remapping them with `Setup Device Inputs`.  When I got to the "click the touchpad" section, it wasn't registering my touchpad clicks.  It never asked me for rear paddles.

I guess I'll have to reboot into Windows and use Legion Space to swap my start+select and Legion buttons.  I've been doing it in Steam Settings with the `xpad` driver, but since the menu button is chorded with `hhd`, it isn't easily swapped in Steam Settings.

Also, I'm still seeing the Legion Controller in Steam Settings in 0.2.6.

appsforartists, 2023-12-30:
Just caught this in the README:

> Another set of obscure issues occur depending on how apps hook to the PS5 controller.
If the PlayStation driver is not active, the Linux kernel creates an evdev node
with incorrect mappings (right trigger becomes a stick, etc).
If the app hooks directly into the hidraw of the controller, it works properly.
If it uses the evdev device its incorrect.

but it's not clear what the resolution is.

antheas, 2023-12-30:
Reference my comments on the nix pull.
https://github.com/NixOS/nixpkgs/pull/277661

Essentially, missing `hid_playstation` + udev rules together.

antheas, 2023-12-30:
I have not installed your linux patch. If that messes with controller hiding/hhd working, open a new issue.

closed 2023-12-30 by antheas

antheas, 2023-12-30:
https://github.com/hhd-dev/hhd/issues/11