hhd-dev/hhd #13: Feature request: ability to select which button gets chorded

appsforartists, 2024-01-03:
When using the DualSense Edge bindings for the Legion Go, every button is accounted for except Legion R, which gets chorded into two buttons. Unfortunately, the chording prevents this key from being remapped in Steam. 

Is it possible to choose which button gets chorded, to allow the others to be mapped in Steam?

antheas, 2024-01-03:
It is in the works to be able to remap any button to any button combination. For now, I have patched together some commonly requested ones

The option you are referring to is called `share_to_qam` in the config.

antheas, 2025-06-30:
Perhaps in the future this will happen outside of hhd, but it is out of scope now.