hhd-dev/hhd #23: Controller hiding breaks external controllers

TaitTechMedia, 2024-01-27:
Is there an easy way to toggle off the controller hiding. As far as I can tell it doesn't seem to be done with a udev rule but it still breaks my Xbox elite controller connected via Bluetooth.

antheas, 2024-01-27:
Controller hiding is done with udev dynamically and only hides the built in controller

You can find the rule in `/run/udev/rules.d` while the controller is hidden.

So can you give me more context?

Device, version, etc. what do you mean breaks?

Also check if there's a conflict with the elite controller and the rule

TaitTechMedia, 2024-01-29:
Interesting, okay so I am on an ROG ally running Nobara on kernel 6.7.2 with a few IMU patches, but nothing that should effect controller visibility.

I am attempting to connect a xbox one elite controller over bluetooth. Connects fine but refuses to appear to the OS in any way.

No UDEV rules currently mention/contain any reference to the xbox controller device id.

Essentially, nothing I do, even deleting the HHD udev rule makes the controller appear, but uninstalling HHD has the controller working again.

I am going to try a few different kernels, to see if this is specific to one kernel.

antheas, 2024-01-29:
Oh, so uninstalling fixes the controller. Can you follow the manual install steps step by step and tell me when the issue appears? My guess is the current udev rules might have a mistake, since that is the only file introduced during install.

antheas, 2024-01-29:
But there is nothing mentioning an xbox elite controller here
https://github.com/hhd-dev/hhd/blob/master/usr/lib/udev/rules.d/83-hhd.rules

Does the controller work when hhd is installed?

TaitTechMedia, 2024-01-29:
Okay so no errors when manually installing and if HHD is installed and disabled, the controller does indeed appear and function correctly. 

I was playing around and found that under desktop mode the controller always appears properly without any issue, however, in opengamepad UI ONLY the controller set in the decky plugin appears. In other words, it does work as intended under desktop mode, but in game mode, it seems to not allow the bluetooth controllers to display properly.

I am thinking maybe this isn't an HHD issue. I am going to nuke the machine with a completely fresh install on a vanilla kernel and work up from there to see if it works and where in the chain it breaks.

antheas, 2024-01-29:
Opengamepadui tries to do some remapping of its own and might be consolidating the controllers. I heard there are issues with external controllers too.

Handheld Daemon has never been tested in it, so reports are welcome. However since Handheld Daemon wraps the controller completely, it should work fine.

I would discourage you reinstalling your system as I doubt that it is the Kernel's fault.

Thanks for the report. You can close it after verifying it is not related to hhd.

TaitTechMedia, 2024-01-29:
Okay confirmed, this is not an issue with HHD but specifically with the Elite Controller and OpenGamepad UI. Dualsense controllers work completely fine as does standard xbox 360 controllers. Apologies for the false issue!

closed 2024-01-29 by TaitTechMedia