hhd-dev/hhd #29: [Feature Request] Allow command line configuration.

dreirund, 2024-02-02:
So far, you provide only a HTTP configuration interface.

`hhd --help` says I can also pass `[command]`, but I find it nowhere documented.

Can you make this configurable also via command line, please?

So far I did not get anything meaningful out of this; I have a GPD Win Max 2 (2023), and `hhd` only loads the plugin `powerbuttond`.

Regards!

antheas, 2024-02-02:
Hi dreirund!

Your device is supported by the latest version, make sure you are using 1.0.4. If it still does not work by default, I would greatly appreciate it if you could follow the steps below and pull request me. Having this information is important for supporting devices.
https://github.com/hhd-dev/hwinfo/tree/master/devices

It is probably due to a failed DMI check:
https://github.com/hhd-dev/hhd/blob/c980f91e6bd01b10b767f13a090ffe4e1fa15ea5/src/hhd/device/gpd/win/__init__.py#L20

Right now, the only devices hhd runs on are devices that we have personally verified they work. I plan for the next version to auto-enable support for GPD and Ayaneo devices even if I have not validated they work using different checks.

Most users like the decky plugin, and a few use hhd.dev, so you can check those out when your device starts working. You are right though, it should have a cli interface, but there are a couple of implementation blocks for that.

For now, if you want to interact with hhd through the command line, you can use curl with the following:
https://github.com/hhd-dev/hhd/blob/master/docs/http.md

It is not the best, I agree.

The only supported command right now is `hhd token`, which reads your user's token.

I will leave the issue open as cli interaction is something I would like to support. You may open another issue about your device not being supported.

dreirund, 2024-02-02:
> *Most users like the decky plugin*

I don't use any steam stuff,

> and a few use hhd.dev,

I now for a test run [`hhd-ui`](https://github.com/hhd-dev/hhd-ui) locally; I want to try to properly package it for Arch Linux but that NPM-stuff is a hell for me and I don't know how to do it properly and clean without added complexity layers like AppImage.

> *I will leave the issue open as cli interaction is something I would like to support.*

Great!,

> *You may open another issue about your device not being supported.*

OK, here:  
#31.

antheas, 2024-02-02:
If you do not use steam, I do not know what the current version of this software will offer you, as remapping is not supported yet. Essentially it would boil down to enabling R4/L4 for use in other software and gyro in yuzu.

dreirund, 2024-02-02:
> If you do not use steam, I do not know what the current version of
> this software will offer you, as remapping is not supported yet.
> Essentially it would boil down to enabling R4/L4 for use in other
> software and gyro in yuzu.

Since I got not really knowledgeable from the documentation I hoped
that with this software I could do some settings that otherwise I can
only do with "Motion Assistant" on Windows (changing power limits etc.)
or re-programming the bottom-side buttons that emit keypress events.

So I wanted to get it functioning to then see what it can do.

Regards!

dreirund, 2024-02-03:
Interesting: [&nearr;&hairsp;Here](https://gitlab.freedesktop.org/drm/amd/-/issues/3073#note_2267198) it is said

> *I have asked GPD officers, -05 is a test BIOS. They don't advice to add -05.*

So `-04` seems to be the default. And `-05` something "for developers"?

toast003, 2024-02-05:
> So `-04` seems to be the default. And `-05` something "for developers"?
05 is for testing BIOS, I somehow had one installed on my wm2 when @antheas and me were getting wm2 support going

toast003, 2024-12-03:
@dreirund hhd 3.6.2 now has hhdctl so you can close this issue now

antheas, 2024-12-03:
I was actually looking for this issue when I made `hhdctl` and missed it.

closed 2024-12-03 by antheas