hhd-dev/hhd #30: Please do _not_ silently execute downloaded stuff!

dreirund, 2024-02-02:
I just played around with [`hhd-ui`](https://github.com/hhd-dev/hhd-ui) and I found a button "update Decky" there.

It apparently did nothing, but on the terminal where `hhd` runs it is printed:

```
Command 'curl -L https://github.com/hhd-dev/hhd-decky/raw/main/update.sh | HOME="/root/" sh' returned non-zero exit status 255.
```

So it actually invokes remote file download and execution as the local user. Without any warning. Without any looking-at.

This is a major security issue:

* There is no guarantee that the retrieved file is what it is intended (malicious third party could intercept).
* It is prone to errors by mistake.

Also, it maybe does stuff on a system even when there is no "Decky" or when the decky plugin is installed via the package manager.

Please remove the option, or move it to an advanced section with an explicit explanation of whait does in the background and a big warning.

Regards!

antheas, 2024-02-02:
Hi dreirund, first off, the command log you posted shows that you did not follow the install instructions. Please do that in the future before opening issues. Handheld Daemon is not bound to your user right now.

I will agree with you that there is a gray area with allowing Handheld Daemon to pull binaries from third party sources when it is installed by a package manager. It was not a priority to fix as Handheld Daemon is not yet in upstream repositories.

However, the URL you posted is secured by HTTPS and on a public known repository which is also write protected to the same authors the PyPi download and the main repository is. This standard is in line with what is used by the majority of software in the market right now. And it only runs when the user presses it. 

The current script does nothing when Decky is not installed in your system. Updating Decky is core functionality right now expected by most users, so it can not be moved to an advanced section.

As for the install instructions themselves, unfortunately this unclean form of installation is what is used for decky plugins.

antheas, 2024-02-02:
Nevertheless, I will think about how to fix it these days.

The above comment is to say that currently there is no severe secure issue that should be immediately fixed.

antheas, 2024-02-06:
The next Decky plugin version bundles the updater in it and hides the current update buttons.

Added deprecation warning to remind users to update.
https://github.com/hhd-dev/hhd/commit/2d2f855c36286653313d43ae2999d4a16ab04184

The button will be removed in a release in a week or two to give people time to update. After that, they will need to use the command line again.

antheas, 2024-02-16:
https://github.com/hhd-dev/hhd/commit/f47c3b8f3803944f31f05021a497cd9c7e813bf0

closed 2024-02-16 by antheas