hhd-dev/hhd #31: GPD Win Max 2 (2023), board name `G1619-04`, not supported.

dreirund, 2024-02-02:
*This is a follow up from [this post](https://github.com/hhd-dev/hhd/issues/29#issuecomment-1923702075) in issue #29:*

---

> *Your device is supported by the latest version, make sure you are using 1.0.4.*

I use latest git checkout, currently at commit c980f91. `git describe --tags` says `v1.1.4-16-gc980f91`. (The release installation from the [`hhd` AUR package](https://aur.archlinux.org/packages/hhd) did not start up at all, so I made a [`hhd-git` package](https://aur.archlinux.org/pkgbase/hhd-git) and used this.)

When I start `hhd` as root, it prints:

```
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Running as root without a specified user (`--user`). Configs will be placed at `/root/.config`.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
   _______   _______    ______  
  /    /  ╲╲/    /  ╲╲_/      ╲╲
 /        //        //        //
/         /         /         / 
╲___/____/╲___/____/╲________/  

[14:15] MAIN  INFO     Handheld Daemon starting...                                                                                                                                                   
        MAIN  INFO     Running autodetection...                                                                                                                                                      
        MAIN  INFO     Found plugin providers: ayaneo, gpd_win, legion_go, powerbuttond, rog_ally                                                                                                    
        MAIN  INFO     Loaded the following plugins:                                                                                                                                                 
                         - powerbuttond: powerbuttond@'uknown'                                                                                                                                       
        PBTN  INFO     Starting isa keyboard powerbutton handler for device 'uknown'.                                                                                                                
        MAIN  INFO     Reloading configuration.                                                                                                                                                      
        PBTN  INFO     Waiting for steam to launch.                                                                                                                                                  
        MAIN  INFO     No profiles found.                                                                                                                                                            
        MAIN  INFO     Initialization Complete!                  
```

> *If it still does not work by default, I would greatly appreciate it if you could follow the steps below*

OK,

> *and pull request me.*

this I don't know how to do and is to complicated for me. I continue to report here, and you can directly include in your repository in a way that suits your coding.

> *Having this information is important for supporting devices. https://github.com/hhd-dev/hwinfo/tree/master/devices*
> 
> *It is probably due to a failed DMI check:*
> 
> *https://github.com/hhd-dev/hhd/blob/c980f91e6bd01b10b767f13a090ffe4e1fa15ea5/src/hhd/device/gpd/win/__init__.py#L20*

On my system, files in `/sys/devices/virtual/dmi/id/` say:

* `/sys/devices/virtual/dmi/id/product_name`: `G1619-04`
* `/sys/devices/virtual/dmi/id/board_name`: `G1619-04`
* `/sys/devices/virtual/dmi/id/uevent`: `MODALIAS=dmi:bvnAmericanMegatrendsInternational,LLC.:bvr0.35:bd09/12/2023:br0.35:efr0.14:svnGPD:pnG1619-04:pvrVer.1.0:rvnGPD:rnG1619-04:rvrVer.1.0:cvnDefaultstring:ct10:cvrDefaultstring:sku7896559068635:`

That is different from your `G1619-05`.

* `/sys/devices/virtual/dmi/id/bios_release`: `0.35`

antheas, 2024-02-02:
You may pull the latest git version and test it
https://github.com/hhd-dev/hhd/commit/55d47848f9bf3e1a49d691699862c2951065ec86

dreirund, 2024-02-02:
> You may pull the latest git version and test it [55d4784](https://github.com/hhd-dev/hhd/commit/55d47848f9bf3e1a49d691699862c2951065ec86)

I think you also should touch `plugins/powerbutton/const.py`, since it also has `G1619-05` defined for the "GPD Win Max 2 2023".

closed 2024-02-02 by antheas