hhd-dev/hhd #37: HHD Block System Poweroff

chenxijun, 2024-02-06:
Device: Ayaneo Air (Standard) 16+512
System: Chimera OS 46 Unstable (SK Modified) [repo](https://github.com/3003n/chimeraos)
Version: newest (ce940eb)

Confirmed it was caused by hhd. It got stuck in such loop:

```
Feb 07 00:51:50 chimeraos kernel: usb 1-4: USB disconnect, device number 87
Feb 07 00:51:49 chimeraos (udev-worker)[4299]: event12: Process '/bin/systemctl start controller-wake@/dev/input/event12.service' failed with exit code 4.
Feb 07 00:51:49 chimeraos systemd[1]: Requested transaction contradicts existing jobs: Transaction for controller-wake@-dev-input-event12.service/start is destructive (veritysetup.target has 'stop' job queued, but 'start' is included in transaction).
Feb 07 00:51:48 chimeraos (udev-worker)[4300]: js0: Process '/bin/systemctl start controller-wake@/dev/input/js0.service' failed with exit code 4.
Feb 07 00:51:48 chimeraos systemd[1]: Requested transaction contradicts existing jobs: Transaction for controller-wake@-dev-input-js0.service/start is destructive (systemd-remount-fs.service has 'stop' job queued, but 'start' is included in transaction).
Feb 07 00:51:48 chimeraos (udev-worker)[4299]: input108: Process '/bin/systemctl start controller-wake@.service' failed with exit code 1.
Feb 07 00:51:48 chimeraos kernel: input: Microsoft X-Box 360 pad as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-4/1-4:1.0/input/input108
Feb 07 00:51:48 chimeraos kernel: usb 1-4: Manufacturer: ZhiXu
Feb 07 00:51:48 chimeraos kernel: usb 1-4: Product: Controller
Feb 07 00:51:48 chimeraos kernel: usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Feb 07 00:51:48 chimeraos kernel: usb 1-4: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.10
Feb 07 00:51:48 chimeraos kernel: usb 1-4: new full-speed USB device number 87 using xhci_hcd
Feb 07 00:51:48 chimeraos kernel: xpad 1-4:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
Feb 07 00:51:48 chimeraos kernel: usb 1-4: USB disconnect, device number 86
```

This situation made hhd not able to gracefully shutdown. It would reach 1min30s timeout of systemd and be killed, then poweroff process continued.

Full log seen here: [gist](https://gist.github.com/chenxijun/4b5c65a67675a2d6d73c974017af8db6)

antheas, 2024-02-06:
Does this happen every time or periodically?

I think I know the issue, but it is hard to diagnose without a device.

antheas, 2024-02-06:
Try pulling from git now. I found a bug related to your device.

You can press update unstable if you have the local install.

antheas, 2024-02-06:
This bug was caused by the powerbutton daemon having a bug and getting deadlocked. Nice catch!
```
Feb 07 00:50:22 chimeraos hhd[543]:         MAIN  INFO     Stopping plugin `powerbuttond@'uknown'`.
```
You can see that it froze after trying to stop the power button daemon. But other than that, since the controller plugin exited, anything else in the logs is probably unrelated to handheld daemon.

However it might be another bug related to controller hiding and ChimeraOS.

antheas, 2024-02-06:
Reference the following script which appears to be causing the error loop.
https://github.com/ChimeraOS/device-quirks/blob/main/usr/bin/controller-wake

chenxijun, 2024-02-07:
> Try pulling from git now. I found a bug related to your device.
> 
> 
> 
> You can press update unstable if you have the local install.

Nice job! The bug is fixed after upgraded. Thank you!

closed 2024-02-07 by chenxijun