hhd-dev/hhd #38: Legion Go : touch screen doesn't work when game are runing

HarAakRanaud, 2024-02-14:
in many games  when i need to use the touch screen (miniMetro, Civilisation 5, VRChat) it doesn't work

sometime a cursor appears but no always i can't click when a game is runing.

i don't know if it's an probleme from HHD or my setup on my the legion go

if it can help i have made a vidéo of the issue
https://youtu.be/2wdDfE5XYbs

antheas, 2024-02-14:
Hi Har,
This is completely unrelated to hhd but I will not close the issue yet.

What os are you running?

The touch gestures are not provided by hhd, but by a modified version of gamescope.

HarAakRanaud, 2024-02-14:
i use NobaraOs

matte-schwartz, 2024-02-14:
> i use NobaraOs

Was it working until recently or has this always not worked for you? The only package updated in the past couple days was Gamescope, so if this just popped up, can you confirm whether or not the problem persists after `sudo dnf downgrade gamescope` to test the previous Gamescope version.

HarAakRanaud, 2024-02-14:
this always not worked, i have try sudo dnf downgrade gamescope but this issue are always here.

matte-schwartz, 2024-02-14:
> this always not worked, i have try sudo dnf downgrade gamescope but this issue are always here.

I don’t have a device to test this stuff myself. Does the touchscreen work as expected if you launch games from desktop? The sliding gestures won’t work but that’s normal.

antheas, 2024-02-14:
It should be the same on all devices. It might be related to the recent gamescope patch for the gestures.

HarAakRanaud, 2024-02-15:
> It should be the same on all devices. It might be related to the recent gamescope patch for the gestures. Or it might be just gamescope being gamescope

i have downgrade the gamescope but the issue is always here

matte-schwartz, 2024-02-15:
It is a problem with the steam input preset you are using. Change to a different controller layout with steam input by toggling “show all devices” and try a community layout made for Steam Deck.

antheas, 2024-03-13:
Problem was found. I'm also getting this bug now, hella annoying.

closed 2024-03-13 by antheas