hhd-dev/hhd #44: Turn On/Off automatically on exit/enter game mode

Tekkenfede, 2024-02-28:
On Gpd win mini I'm obliged to open the UI and disable emulation(decky or desktop) to make work again the touchpad in desktop mode with gestures(that on a similar device means using it like a normal device)

Is it eventually possible to add a toggle to make possible an automatic on/off like said in the title?
Cause in gamemode having touchpad emulation (controller or virtual) is the only way to have an effective left click from the touchpad.(while it messes up things on the desktop)

Ps: I'm on bazzite

antheas, 2024-02-28:
I hope to to this as part of per-game profiles but I will try to think of a way to do this before hand.

The issue right now is that I dont know if you are in gamemode or not but that will change soon.

It will just take a bit of time. Doesn't the Win Mini have a touchpad click?

antheas, 2024-02-28:
Yep, just wondering if you could go by by disabling it all the time

But now I recall that the win max is the one that also clicks

Tekkenfede, 2024-02-28:
Yeah, in both cases I can open and use without any problem the electron UI to disable it.
Or just use controller commands/touch in decky.
It's just the classic "feature" I would like to see automated if the user wants to.(no hurry/priority, just my 2 cents on a niche device ahah)

antheas, 2024-04-05:
https://github.com/hhd-dev/hhd/releases/tag/v2.5.0
https://github.com/hhd-dev/hhd/commit/6cb10213cdd9ae465694b7234cf8b2cf4c736016

Enjoy.

closed 2024-04-05 by antheas

Tekkenfede, 2024-04-09:
Tried today! Thanks a lot, it also seems to disable/enable on the go if you set steam input in desktop mode, something really cool.(Dunno if expected, but in the worst case, nice bug ahaha)