hhd-dev/hhd #51: Controller Emulation breaks after sleep on Aokzoe A1

byronpc, 2024-04-09:
I recently installed BazziteOS with HHD-decky

My device is Aokzoe A1 

Upon restart HHD controller emulation works (e.g. dualsense) which makes the home and quick menu button work. However upon sleep and wake up of the device, the controller reverts to Xbox 360 controller that doesn't have support of the 2 buttons.

Any help would be very much appreciated

antheas, 2024-04-09:
Hi Byron, I hope the bazzite ver auto started for your device and you didn't try to use the install script of this repository 

Can you look in journalctl for logs during the suspend?

`journalctl | fpaste`

I made some changes to device code recently to make reconnections faster and maybe there's a slight issue with yours.

byronpc, 2024-04-09:
https://paste.centos.org/view/f7587a3d

antheas, 2024-04-09:
Indeed, this was the case.
https://github.com/hhd-dev/hhd/commit/f3a8913e1cda416841897b5e10f14a8ed095b351

I missed a variable. This patch will be out today or tomorrow.

Until then, when you boot, change a controller setting in hhd (such as going to xbox and back to dualsense). This will skip the crash until you reboot.

Thanks for reporting.

closed 2024-04-09 by antheas

byronpc, 2024-04-11:
btw i saw that you've already released the hotfix for this yesterday. any idea how to get this update on bazzite?