hhd-dev/hhd #52: [NixOS] README might be slightly wrong on the process to get HHD working on NixOS?

hanakomisa, 2024-04-12:
In the README it mentioned adding `services.handheld=daemon.enable = true;` to your `configuration.nix`, however, upon doing `nixos-rebuiid`, the following error is thrown out:
```
building the system configuration...
error:
       … while calling the 'head' builtin

         at /nix/store/r1g09wfd968168f6drdl7agfaf3msik4-source/lib/attrsets.nix:1541:11:

         1540|         || pred here (elemAt values 1) (head values) then
         1541|           head values
             |           ^
         1542|         else

       … while evaluating the attribute 'value'

         at /nix/store/r1g09wfd968168f6drdl7agfaf3msik4-source/lib/modules.nix:809:9:

          808|     in warnDeprecation opt //
          809|       { value = builtins.addErrorContext "while evaluating the option `${showOption loc}':" value;
             |         ^
          810|         inherit (res.defsFinal') highestPrio;

       (stack trace truncated; use '--show-trace' to show the full trace)

       error: The option `services.handheld-daemon.user' is used but not defined.
```

I decided to add `services.handheld-daemon.user = "USERNAME";` per the error's suggestion and I was able to get a successful rebuild, alongside being able to connect to my HHD service through https://hhd.dev as well.

Is this just an edge case of my install being started out from 23.11 then upgraded to unstable, or is the instruction outdated?
Let me know if you need more info.
-Neon

antheas, 2024-04-13:
Hi,
Yes both the instructions and the package on nix are outdated. I'm working on a new readme.

New ver is 2.6.2 and I think nix has an early 1.#.# variant and misses the overlay and tdp

hanakomisa, 2024-04-13:
Aha, thanks for the response! And yes the reported version for me was 1.1.0, no TDP control and no overlay, though my config is also still a big WIP since the Ally is definitely a unicorn when it comes to Linux in general. Thanks for the work that you all do to make our handhelds work and work well.

antheas, 2024-04-26:
Fixed and nix is updated now. At least I think. I have not tested the config parameters.
https://github.com/hhd-dev/hhd#%EF%B8%8F-nixos