hhd-dev/hhd #55: [Bug] Bazzite Legion Go - Home / QAM input delay only applies to Legion Button inputs

Shaunathanz, 2024-04-21:
Handheld Daemon Version: 2.5.3
Handheld Daemon UI Version: 2.2.1
Adjustor (TDP) Version: 2.1.3
HHD Decky Version: 0.1.0
Tested on various recent releases of Bazzite OS based on Fedora 39.

So I have tried multiple different Bazzite releases and a few different versions of HHD that I know used to work and it didn't fix my issue.

The Legion L and R buttons on the True Strike controllers can be mapped to either Home / QAM access buttons or swapped to be Select / Start buttons. I started encountering an issue after some recent updates and maybe a hard reset or two where pressing Legion L / R buttons causes all other controller inputs to not be detected for a good half second or so. I have a feeling this input nullification is meant for the Home / QAM menu navigation, but when I am using those buttons as Start / Select I don't want my analog inputs or held buttons to be ignored or interrupted. At the same time the two buttons that act as Start / Select by default have no input nullification, inputs are not interrupted even when those buttons are mapped to Home / QAM.

I tried deleting every HHD folder within the home directory to no avail, rebasing hasn't helped, downgrading HHD hasn't helped, reinstalling the decky plugin, toggling on and off all settings hasn't helped, also tried factory resetting controllers to no avail. Not sure where else the issue could be but hopefully someone here may have an idea.

antheas, 2024-04-21:
How did you swap the legion buttons?

Shaunathanz, 2024-04-21:
I think I'll try a clean reinstall of the entire OS before I reopen

antheas, 2024-04-21:
That happens if you use the legion remapper plugin instead of the hhd option

Input nullification is a built in feature of the controllers that is partly undone to allow chording. It breaks if you use the legion space option that is available through legion remapper

Shaunathanz, 2024-04-21:
> How did you swap the legion buttons?

The Swap Legion with Start/Select option has been toggled for quite a while now. A recent update to Bazzite introduced some suspend issues so I rebased to an older release. Once I loaded up the OS again I noticed the issues I described above. I tried rebasing a few more times (all rebases are Bazzite GNOME) and the behavior is exactly the same as far as these inputs are concerned. Since I deleted all the HHD folders and the issue still persisted I can only guess something beyond the scope of Bazzite and HHD must be corrupted, because nothing I do seems to fix it. 

Before I rebased to fix the suspend issue I noticed that detaching at least one True Strike controller would allow suspend to work again, but I don't know enough about HHD or Bazzite's inner workings to be able to identify any significance in that observation.

antheas, 2024-04-21:
Where did you find that option?

antheas, 2024-04-21:
With the handheld daemon option. Holding a button combo is not the same.

Shaunathanz, 2024-04-21:
> Where did you find that option?
> 
> Oh decky plugin. Try factory resetting the controllers again

Not sure if factory reset behaves differently when used wirelessly or connected, but I did reset them with the factory reset option in Handheld Daemon's Decky plugin and it didn't seem to affect anything. I also uninstalled Legion go remapper and restarted before trying to reduce any potential interference from that plugin

antheas, 2024-04-21:
Try it through the overlay by double pressing legion R. The led colors should switch to green for a second.

Shaunathanz, 2024-04-21:
No luck. New overlay toggles work the same as the Decky plugin toggles and I just can't get that input nullification to transfer with the button remaps that HHD allows, they just stick with Legion L and R. I restarted after disabling and only made the change with the new overlay but same result. Would prefer not to wipe the OS so if you have any other troubleshooting ideas please let me know

antheas, 2024-04-21:
Oh I get what you mean. I can see it happening. I don't think an update caused it I think it was either always there or part of the new firmware. In any case, don't format your device.

Shaunathanz, 2024-04-21:
I'll hold off on the format. I think I should probably mention that I don't boot into Windows so my controller firmware and BIOS are a bit outdated, not sure if these plugins are backwards compatible with older firmwares or if the latest is a requirement.

antheas, 2024-04-21:
I have not dealt with the nullification in the case you release the buttons. You're the first one to notice

BIOS updates are important as TDP controls were broken a couple of updates ago. Controller firmware not so much. There hasn't been a new worthwhile addition to it as far as Linux is concerned 

As for the suspend issue, roll back to the 0406 image

Shaunathanz, 2024-04-21:
If I recall correctly I'm on V28 or 29, whichever the first one was that fixed custom TDP mode, and I actually did go with the 04-06 image :)

As for the input nullification I believe the actual input event that triggers it is the key down, because if I hold the Legion button the controls come back after ~ 1 second. So if I hold A, then hold a Legion button down, after a small wait they will both register as held down.

antheas, 2024-04-21:
I will need to rework that in the future 

As for the bios. The only notable change in newer bioses is fixing the fan curves

Shaunathanz, 2024-04-21:
All your work is very much appreciated, thanks for responding so quickly.

Shaunathanz, 2024-05-05:
Reopening to report issue for HHD 2.7.1

antheas, 2024-05-05:
Can you try master?

antheas, 2024-05-05:
Also can you verify by going I to hhd settings the correct version is running, that is 2.7.1. since bazzite launches the built in service every boot

antheas, 2024-05-05:
So I just tested version 2.7.1 as packaged for arch and the legion buttons work great without nullification.

You can always just download and run the newer version a bit, it just wont stick properly after reboots.

As part of the factory firmware, after you press the legion buttons, it nullifies all other buttons for 1 second. I try to get around that by splicing the inputs from another device. On the new version I always do it for 1 second and it works perfectly on my machine.

Shaunathanz, 2024-05-05:
So I guess I'm doing something wrong. I commented out the Bazzite check in install.sh and installed it. Saw the notice about removing bundled HHD, so I run systemctl disable --now hhd@(my user) and systemctl daemon-reload and systemctl restart plugin_loader.service. I'm not seeing a new HHD process running so I assume I'm missing something obvious. Reboot didn't help either, no HHD running.

antheas, 2024-05-05:
Yes, run the uninstall script then enable the built in service. You can use the script below to enable a different hhd version until your reboot.
```bash
sudo systemctl stop hhd@$(whoami) 
sudo pkill hhd       

rm -rf ~/.local/share/hhd
mkdir -p ~/.local/share/hhd
python -m venv --system-site-packages ~/.local/share/hhd/venv
~/.local/share/hhd/venv/bin/pip install git+https://github.com/hhd-dev/adjustor git+https://github.com/hhd-dev/hhd

FINAL_URL='https://api.github.com/repos/hhd-dev/hhd-ui/releases/latest'
curl -L $(curl -s "${FINAL_URL}" | grep "browser_download_url" | cut -d '"' -f 4) -o $HOME/.local/share/hhd/hhd-ui
chmod +x $HOME/.local/share/hhd/hhd-ui

nohup sudo HHD_OVERLAY="$HOME/.local/share/hhd/hhd-ui" ~/.local/share/hhd/venv/bin/hhd --user $(whoami) &> /dev/null &
```

Shaunathanz, 2024-05-05:
[image]
After reenabling HHD, uninstalling the hhd_local files, running the script you commented, and switching back to game mode it appears that HHD is now at 2.7.0 and the issue persists. I checked with the overlay and did a couple factory resets on the controllers and the behavior is the same. I've even changed the emulated devices with no success. I am emulating the Xbox controller.

antheas, 2024-05-05:
It works perfectly on my end. And you dont use the remapper for legion swap you use hhd. Weird, maybe its time for a firmware update?

Shaunathanz, 2024-05-05:
I was thinking the same, firmware update is the only thing I haven't tried 🤣

Shaunathanz, 2024-05-05:
Updated BIOS and controller firmware, same exact behavior. 

I guess reinstalling the operating system really may be the only answer

antheas, 2024-05-05:
Yeah it's in ~/.config/hhd

antheas, 2024-05-05:
A video of the steam controller tester would be nice so I understand what's the problem

Shaunathanz, 2024-05-05:
https://quickshare.samsungcloud.com/fCnBuqMcCEKW

antheas, 2024-05-05:
Oh the dpad is broken, you could have been a bit more explicit about it. All other buttons work. Guess for fighting games that's not ideal, I will work on fixing that.

Yes the buttons do reset when you press the button, I can work on that but I cant promise a solution since it will take quite a bit of a rewrite and I don't think it affects functionality.

Shaunathanz, 2024-05-05:
Yeah I only just realized the dpad had the 1s no input ussue while recording it. I did update the previous comment a bit but yeah the issues are basically the dpad thing and legion inputs resetting the state of all controller inputs when pressed. Sorry to bug you with so many legion-specific problems

antheas, 2024-05-05:
https://github.com/hhd-dev/hhd/releases/tag/v2.7.2
https://github.com/hhd-dev/hhd/commit/8dd5c633c3b851e5cacd4ff0f240c18b60afb939

I will keep in mind the outputs being cancelled and maybe try to work around it in the future. However, chording a button and then pressing the modifier, e.g., A + start/select/share/guide is not something that is done, so you can only tell this is an issue on the controller tester.

If the issue persists you can reopen the issue.

closed 2024-05-05 by antheas