hhd-dev/hhd #59: Suspend reverts to native controller on Ally

acuteaura, 2024-04-26:
I'm seeing #51 the Ally, always reproducible after suspend, running Bazzite. Notably changing device emulation does not fix it here though. I have to `systemctl restart hhd@<username>`.

```
<resume at this exact moment>
Apr 26 11:43:22 ally hhd[10125]:         ALLY  ERROR    Received the following error:
Apr 26 11:43:22 ally hhd[10125]:                        <class 'OSError'>: [Errno 19] No such device
Apr 26 11:43:22 ally hhd[10125]:         ALLY  ERROR    Assuming controllers disconnected, restarting after 0.3s.
Apr 26 11:43:22 ally hhd[10125]: Exception in thread Thread-2 (plugin_run):
Apr 26 11:43:22 ally hhd[10125]: Traceback (most recent call last):
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib64/python3.12/threading.py", line 1073, in _bootstrap_inner
Apr 26 11:43:22 ally hhd[10125]:     self.run()
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib64/python3.12/threading.py", line 1010, in run
Apr 26 11:43:22 ally hhd[10125]:     self._target(*self._args, **self._kwargs)
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib/python3.12/site-packages/hhd/device/rog_ally/base.py", line 169, in plugin_run
Apr 26 11:43:22 ally hhd[10125]:     raise e
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib/python3.12/site-packages/hhd/device/rog_ally/base.py", line 154, in plugin_run
Apr 26 11:43:22 ally hhd[10125]:     controller_loop(conf.copy(), should_exit, updated, emit)
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib/python3.12/site-packages/hhd/device/rog_ally/base.py", line 268, in controller_loop
Apr 26 11:43:22 ally hhd[10125]:     evs.extend(d.produce(r))
Apr 26 11:43:22 ally hhd[10125]:                ^^^^^^^^^^^^
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib/python3.12/site-packages/hhd/controller/physical/evdev.py", line 377, in produce
Apr 26 11:43:22 ally hhd[10125]:     for e in self.dev.read():
Apr 26 11:43:22 ally hhd[10125]:   File "/usr/lib64/python3.12/site-packages/evdev/eventio.py", line 71, in read
Apr 26 11:43:22 ally hhd[10125]:     events = _input.device_read_many(self.fd)
Apr 26 11:43:22 ally hhd[10125]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Apr 26 11:43:22 ally hhd[10125]: OSError: [Errno 19] No such device
```

antheas, 2024-04-26:
Did you enable the debug option in hhd settings? If yes remove it.

I will also try to remove it in the next ver, it's not the first time it turned on on its own

acuteaura, 2024-04-26:
Thanks for the heads up, that indeed was the problem, could've been me turning that on debugging other issues.

closed 2024-04-26 by acuteaura