hhd-dev/hhd #65: [Legion Go] Controller input is ignored for a second after pressing "Legion R" button

mvasl, 2024-05-03:
https://github.com/hhd-dev/hhd/assets/5184586/c3b1cf2d-6a8e-4ea9-94c3-106f1167c8f0

antheas, 2024-05-03:
Thanks for reminding me, this should fix this.
https://github.com/hhd-dev/hhd/commit/e26491843ca727122c4daee9b893665eec9883fc
https://github.com/hhd-dev/hhd/releases/tag/v2.6.6

If it still does not work properly you can reopen the issue

closed 2024-05-03 by antheas

mvasl, 2024-05-03:
I am on bazzite and I see a CI job that was supposed to build and push 2.6.6 to copr, but there is only 2.6.5 and then immediately 2.7.0

https://copr-dist-git.fedorainfracloud.org/cgit/hhd-dev/hhd/hhd.git/commit/?id=4565bcd62db78498393846116c874b989c599d4f

antheas, 2024-05-03:
Indeed, well, version bumped the bump. Although 2.7 works fine, its not quite ready yet.
https://github.com/hhd-dev/hhd/releases/tag/v2.7.1

Shaunathanz, 2024-05-05:
I did a factory reset but still seeing the issue on HHD 2.7.1

antheas, 2024-05-05:
Master branch should be fixed. I can try the stable branch maybe I did something wrong.

mvasl, 2024-05-05:
Tested 2.7.1, the issue is still there

antheas, 2024-05-05:
Indeed it was, but only for the d-pad. See here on how to test the newer version on Bazzite+details. https://github.com/hhd-dev/hhd/issues/55

If the issue still persists, I can re-open the issue.