hhd-dev/hhd #66: GPD Win 3

DocMAX, 2024-05-05:
Can you add GPD Win 3 support?

antheas, 2024-05-05:
I would need help with reverse engineering it. Does the win app for remapping the 2 extra buttons work in windows?

DocMAX, 2024-05-05:
Sorry, i'm on Linux only...

antheas, 2024-05-05:
I assume if you bought the device you'd know if it works or not. You don't need to use it.

Does the device have 2 extra buttons?

antheas, 2024-05-05:
Seems like the device has no extra buttons, the most I can do for it is gyro. Otherwise it should fully work without hhd.

Post the following:
```bash
for f in /sys/class/dmi/id/*; do echo "$f:"; sudo cat $f; done > dmi.txt
sudo dmesg > dmesg.txt
ls /sys/bus/iio/devices/
```
Sanitize where appropriate by removing serial numbers, if you care for that sort of thing.

TDP controls for intel devices are not planned at the moment.

DocMAX, 2024-05-05:
[dmesg.txt](https://github.com/hhd-dev/hhd/files/15214276/dmesg.txt)
[dmi.txt](https://github.com/hhd-dev/hhd/files/15214277/dmi.txt)
ls: cannot access '/sys/bus/iio/devices/': No such file or directory

antheas, 2024-05-05:
https://github.com/hhd-dev/hwinfo/tree/master/devices#acpi-collection

Your device does not load a IMU driver right now, I need to look at the ACPI tables to see what they put in there, if they put something in there. I only need the ACPI section, you can skip WMI and below

DocMAX, 2024-05-05:
I think the Win 3 has no gyro

antheas, 2024-05-05:
Hope dies last but you are indeed correct. Does not seem to have Intel ISH (not that I have added support for it) or an off the shelf IMU unit. 

That means that i can only do the paddles for it, and for those see here:
https://github.com/hhd-dev/hhd/commit/ea9119734ffc5ec63ff359dea50af1da5a920bb0

antheas, 2024-05-18:
Hopefully it at least partly worked

closed 2024-05-18 by antheas