hhd-dev/hhd #67: GPD Win 4 Gyro not working

DocMAX, 2024-05-05:
Isn't it supposed to work? In Steam the Gryo test doesn't do anything. PS5 Controller is detected, though.

[image]

antheas, 2024-05-05:
Post your dmesg logs from this boot

antheas, 2024-05-05:
Also make sure you have the dkms package for gyro for your device for arch https://aur.archlinux.org/packages/bmi260-dkms

DocMAX, 2024-05-05:
Ah, bmi260 did it! I used https://aur.archlinux.org/packages/bmi160-aya-neo-dkms-git before.
Thanks!

closed 2024-05-05 by DocMAX

antheas, 2024-05-05:
For anyone referencing this in the future, the bmi160 fix applies to 1-2 obscure 2022 devices that have a weird ACPI entry and the fix is part of the Linux kernel as of 6.9.

That dkms package may not work though.

Over 90% of devices use a bmi260, and the kernel module is not submitted to the kernel yet.

Then there are two devices that use the bmi323, the Ally and an ayaneo device. Support for those is added as of 6.8 or 6.9 I want to say.