hhd-dev/hhd #70: Elite Controller Support

philipobenito, 2024-05-21:
Hi, I'm running Bazzite on Legion Go and wondering how I can emulate the Elite controller, when I choose Xbox it seems to be emulating a standard controller with no back button mapping.

I'm aware I can just use duelsense edge, but I don't use gyro, and prefer the virtual touchpad to the game one. Also, some games force playstation glyphs with it.

It also says in the readme that elite emulation is supported so just wondering if there are extra steps that I haven't found?

antheas, 2024-05-21:
You are indeed right. I got part through the testing and it works great, but did not end up publishing the release because other features such as Joycon emulation needed more work (since it relies on an obscure kernel patch). It is a bit hard to use the git version on Bazzite though.

I will close the issue once I publish the release.

antheas, 2024-06-27:
https://github.com/hhd-dev/hhd/releases/tag/v3.0.0

closed 2024-06-27 by antheas