hhd-dev/hhd #72: Powerbutton enabled but still shutting down

DocMAX, 2024-06-16:
In HHD decky powerbutton is enable, but my Steamdeck still does shutdown.

antheas, 2024-06-16:
Steamdeck? I disabled it on the steam deck because it breaks the volume buttons

You're supposed to inhibit with systemd and use the valve power button daemon

The gamescope session has the commands

antheas, 2024-07-08:
Hopefully you figured it out.

closed 2024-07-08 by antheas

DocMAX, 2024-10-18:
Without this setting in logind.conf the power button still is shutdown down the system (Steam Deck OLED+LCD, NOT GPD WIN 4).
So not sure if this is still a hhd problem...
```
[Login]
HandlePowerKey=ignore
```

antheas, 2024-10-19:
Your session is broken. It needs to start Valve's power button daemon but it does not.

Do not enable the hhd one. It breaks the volume buttons.

DocMAX, 2024-10-19:
Ok strange. With my config all works now. Also the volume buttons. No power button daemon installed.

antheas, 2024-10-19:
The power button of the steamdeck comes out of the AT keyboard and you need Valve's powerbutton daemon to handle it properly + systemd to ignore it.