hhd-dev/hhd #73: [Feature request] Support 3rd party controllers

yoyossef, 2024-06-19:
First of all thank you for doing all of this, it looks like a really neat project, I'm soon receiving my GPD Win Max 2 and I can't wait to use this in gamemode.

Unfortunately exposing gyro, and paddle inputs across Steam is not only an issue on the handhelds you guys support.
Controllers like the Flydigi Apex 4 have 4 paddles, and gyro support that can't be exploited on Linux under gamemode. Not only because the drivers aren't supporting everything but also because Steam isn't letting us exploiting it.

I know this is not necessarily the scope of the project but the aim goal seem really close to what I want to achieve, I'd imagine the drivers for all the handhelds don't support everything all at once, so fixing all of this in the userspace + mocking it as a Dualsense Edge and/or Xbox Elite controller is a superb solution.

Would you think it'd be possible? Any hints on where I should start looking to expose my controller to hhd to enable controller mocks?

Thanks.

antheas, 2024-06-19:
Check your controllers with `hhd.contrib hidraw` to see if you can see the gyro or paddles somewhere. Also try all different modes they have. For gyro, there's usually a switch mode that is switch only but you can trick it into activating.

yoyossef, 2024-06-19:
@antheas So the Apex 4 has many modes, let's focus on the ones available when the controller is connected wired via USB: Switch and PC: XInput, DInput 

### Switch
<details>
<summary>hhd.contrib hidraw</summary>

```
/dev/hidraw8    057e:2009: Usage Page: 0x0009 Usage: 0x0001 Names 'Nintendo Co., Ltd.': 'Pro Controller'
[...]
   387:   3.0961 30ce91008000f11780f1178009dbfffafff10efbfffeff0200dbfffaffff0efaff0000fefff7fff3ff100ffbff0100ffff000000000000000000000000000000
   388:   3.1040 30d091008000f00780f007800a0400f9ff0f0ffdff0300ffff0700faff020ffaff00000300dbfffafff10efbfffeff0200000000000000000000000000000000
[...]
```
</details>
=> So many events captured maybe because of the gyro, the poll rate seems really high. This mode on Steam is recognized as an actual Switch controller and the Gyro functions is working.

<details>
<summary>hhd.contrib evdev</summary>

```
device /dev/input/event25, name "Nintendo Co., Ltd. Pro Controller", phys "usb-0000:64:00.3-1.3.2.3/input0"

{('EV_SYN', 0): [('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)], ('EV_KEY', 1): [(['BTN_A', 'BTN_GAMEPAD', 'BTN_SOUTH'], 304), (['BTN_B', 'BTN_EAST'], 305), (['BTN_NORTH', 'BTN_X'], 307), (['BTN_WEST', 'BTN_Y'], 308), ('BTN_Z', 309), ('BTN_TL', 310), ('BTN_TR', 311), ('BTN_TL2', 312), ('BTN_TR2', 313), ('BTN_SELECT', 314), ('BTN_START', 315), ('BTN_MODE', 316), ('BTN_THUMBL', 317), ('BTN_THUMBR', 318)], ('EV_ABS', 3): [(('ABS_X', 0), AbsInfo(value=-175, min=-32767, max=32767, fuzz=250, flat=500, resolution=0)), (('ABS_Y', 1), AbsInfo(value=0, min=-32767, max=32767, fuzz=250, flat=500, resolution=0)), (('ABS_RX', 3), AbsInfo(value=-175, min=-32767, max=32767, fuzz=250, flat=500, resolution=0)), (('ABS_RY', 4), AbsInfo(value=0, min=-32767, max=32767, fuzz=250, flat=500, resolution=0)), (('ABS_HAT0X', 16), AbsInfo(value=0, min=-1, max=1, fuzz=0, flat=0, resolution=0)), (('ABS_HAT0Y', 17), AbsInfo(value=0, min=-1, max=1, fuzz=0, flat=0, resolution=0))], ('EV_FF', 21): [(['FF_EFFECT_MIN', 'FF_RUMBLE'], 80), ('FF_PERIODIC', 81), (['FF_SQUARE', 'FF_WAVEFORM_MIN'], 88), ('FF_TRIANGLE', 89), ('FF_SINE', 90), (['FF_GAIN', 'FF_MAX_EFFECTS'], 96)]}

```

</details>
=> No paddle support but an extra buttons seems to work (309 button "BTN_Z")

### XInput
#### hhd.contrib hidraw
No entry found so couldn't capture anything.

<details>
<summary>hhd.contrib evdev</summary>

```
└─╼ hhd.contrib evdev
Available Devices with the Current Permissions
device /dev/input/event25, name "Microsoft X-Box 360 pad", phys "usb-0000:64:00.3-1.3.2.3/input0"
Capabilities
{('EV_SYN', 0): [('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 21)], ('EV_KEY', 1): [(['BTN_A', 'BTN_GAMEPAD', 'BTN_SOUTH'], 304), (['BTN_B', 'BTN_EAST'], 305), (['BTN_NORTH', 'BTN_X'], 307), (['BTN_WEST', 'BTN_Y'], 308), ('BTN_TL', 310), ('BTN_TR', 311), ('BTN_SELECT', 314), ('BTN_START', 315), ('BTN_MODE', 316), ('BTN_THUMBL', 317), ('BTN_THUMBR', 318)], ('EV_ABS', 3): [(('ABS_X', 0), AbsInfo(value=6, min=-32768, max=32767, fuzz=16, flat=128, resolution=0)), (('ABS_Y', 1), AbsInfo(value=5, min=-32768, max=32767, fuzz=16, flat=128, resolution=0)), (('ABS_Z', 2), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_RX', 3), AbsInfo(value=6, min=-32768, max=32767, fuzz=16, flat=128, resolution=0)), (('ABS_RY', 4), AbsInfo(value=-7, min=-32768, max=32767, fuzz=16, flat=128, resolution=0)), (('ABS_RZ', 5), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=0, resolution=0)), (('ABS_HAT0X', 16), AbsInfo(value=0, min=-1, max=1, fuzz=0, flat=0, resolution=0)), (('ABS_HAT0Y', 17), AbsInfo(value=0, min=-1, max=1, fuzz=0, flat=0, resolution=0))], ('EV_FF', 21): [(['FF_EFFECT_MIN', 'FF_RUMBLE'], 80), ('FF_PERIODIC', 81), (['FF_SQUARE', 'FF_WAVEFORM_MIN'], 88), ('FF_TRIANGLE', 89), ('FF_SINE', 90), (['FF_GAIN', 'FF_MAX_EFFECTS'], 96)]}
```

</details>
=> As expected only the basic Xbox 360 buttons events are seen

### dInput
<details>
<summary>hhd.contrib hidraw</summary>

```
/dev/hidraw8    04b4:2412: Usage Page: 0x0001 Usage: 0x0005 Names 'Flydigi': 'Flydigi APEX 4'
/dev/hidraw9    04b4:2412: Usage Page: 0x0001 Usage: 0x0001 Names 'Flydigi': 'Flydigi APEX 4'
/dev/hidraw10   04b4:2412: Usage Page: 0xffa0 Usage: 0x0001 Names 'Flydigi': 'Flydigi APEX 4'
/dev/hidraw11   04b4:2412: Usage Page: 0xffef Usage: 0x0000 Names 'Flydigi': 'Flydigi APEX 4'

Enter device path (/dev/hidraw# or #): 8

Traceback (most recent call last):
  File "/usr/bin/hhd.contrib", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/usr/lib/python3.12/site-packages/hhd/contrib/main.py", line 27, in main
    hidraw()
  File "/usr/lib/python3.12/site-packages/hhd/contrib/dev.py", line 80, in hidraw
    d = Device(path=sel)
        ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/hhd/controller/lib/hid.py", line 193, in __init__
    raise HIDException("unable to open device")
hhd.controller.lib.hid.HIDException: unable to open device
```

</details>
=> None of the 4 devices can be read in hidraw for some reason.

<details>
<summary>hhd.contrib evdev</summary>

```
└─╼ hhd.contrib evdev
Available Devices with the Current Permissions
device /dev/input/event25, name "Flydigi Flydigi APEX 4", phys "usb-0000:64:00.3-1.3.2.3/input0"
Capabilities
{('EV_SYN', 0): [('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_DROPPED', 3), ('?', 4)], ('EV_KEY', 1): [(['BTN_A', 'BTN_GAMEPAD', 'BTN_SOUTH'], 304), (['BTN_B', 'BTN_EAST'], 305), ('BTN_C', 306), (['BTN_NORTH', 'BTN_X'], 307 [... cut ...] at=15, resolution=0)), (('ABS_GAS', 9), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=15, resolution=0)), (('ABS_BRAKE', 10), AbsInfo(value=0, min=0, max=255, fuzz=0, flat=15, resolution=0)), (('ABS_HAT0X', 16), AbsInfo(value=0, min=-1, max=1, fuzz=0, flat=0, resolution=0)), (('ABS_HAT0Y', 17), AbsInfo(value=0, min=-1, max=1, fuzz=0, flat=0, resolution=0))], ('EV_MSC', 4): [('MSC_SCAN', 4)]}
```
</details>
=> All the extra buttons seem to work, BTN Z enables a mouse input controlled by the Gyro.

antheas, 2024-06-19:
Xinput does not appear in hidraw and only has the xbox 360 buttons. There is a chance there would be a manufacturer endpoint that shows the extra buttons in hidraw. Are you sure there arent any hidraw pages for xinput? 

Dinput appears in hidraw in general. It should only have one node, so thats the only interesting one really.

Try running the command with sudo. Only controllers with rules in linux will work without.

As for switch, its good that gyro works out of the box. But if there are no manufacturer interfaces it will not work.

In general, usage pages starting with 0xff are vendor specific. If the company implements a controller tester for it, it's very likely it will list the buttons there for the tester to work.

yoyossef, 2024-06-19:
@antheas
> Are you sure there arent any hidraw pages for xinput?

Yes, even tried with sudo, those are the only pages I see when in xinput:
```
Available Devices with the Current Permissions
/dev/hidraw0    1ea7:0064: Usage Page: 0x0001 Usage: 0x0001 Names 'None': '2.4G Mouse'
/dev/hidraw1    04f3:3231: Usage Page: 0x000d Usage: 0x0022 Names '': 'ELAN06A0:00 04F3:3231'
/dev/hidraw2    056a:52e6: Usage Page: 0x0001 Usage: 0x0001 Names '': 'WACF2200:00 056A:52E6'
/dev/hidraw3    cdcd:8484: Usage Page: 0x0001 Usage: 0x0006 Names 'xiudi': 'XD84'
/dev/hidraw4    cdcd:8484: Usage Page: 0x0001 Usage: 0x0001 Names 'xiudi': 'XD84'
/dev/hidraw5    cdcd:8484: Usage Page: 0x000c Usage: 0x0001 Names 'xiudi': 'XD84'
/dev/hidraw6    cdcd:8484: Usage Page: 0xff31 Usage: 0x0074 Names 'xiudi': 'XD84'
/dev/hidraw7    cdcd:8484: Usage Page: 0x0001 Usage: 0x0006 Names 'xiudi': 'XD84'
``` 

> Dinput appears in hidraw in general. It should only have one node, so thats the only interesting one really.
>
> Try running the command with sudo. Only controllers with rules in linux will work without.

Okay it works with sudo, I'll put the output for each of the following pages:

```
/dev/hidraw8    04b4:2412: Usage Page: 0x0001 Usage: 0x0005 Names 'Flydigi': 'Flydigi APEX 4'
/dev/hidraw9    04b4:2412: Usage Page: 0x0001 Usage: 0x0001 Names 'Flydigi': 'Flydigi APEX 4'
/dev/hidraw10   04b4:2412: Usage Page: 0xffa0 Usage: 0x0001 Names 'Flydigi': 'Flydigi APEX 4'
/dev/hidraw11   04b4:2412: Usage Page: 0xffef Usage: 0x0000 Names 'Flydigi': 'Flydigi APEX 4'
```

<details>
<summary>hidraw8</summary>

```
11663:  24.3936 7f7c7f7f8002d80000
 11664:  24.3956 7f7c7f7f8002d80000
 11665:  24.3977 7f7c7f7f8002d80000
 11666:  24.3996 7f7c7f7f8002d80000
 11667:  24.4016 7f7c7f7f8002d80000
 11668:  24.4037 7f7c7f7f8002d80000
 11669:  24.4056 7f7c7f7f8002d80000
 11670:  24.4076 7f7c7f7f8002d80000
 11671:  24.4096 7f7c7f7f8002d80000
 11672:  24.4116 7f7c7f7f8002d80000
 11673:  24.4136 7f7c7f7f8002d80000
 11674:  24.4156 7f7c7f7f8002d80000
 11675:  24.4176 7f7c7f7f8002d80000
 11676:  24.4197 7f7c7f7f8002d80000
```

</details>
=> This is the page for the regular inputs + extra buttons to be read, all of them affect the value, the Z BTN is a toggle and does not reset its state as it enables the weird gyro mouse thing.

<details>
<summary>hidraw9</summary>

```
  5893:  52.5788 02800290ff0000
  5894:  52.5808 02800290ff0000
  5895:  52.5828 02800290ff0000
  5896:  52.5848 02800290ff0000
  5897:  52.5869 028001b0ff0000
  5898:  52.5888 028001b0ff0000
  5899:  52.5908 028001b0ff0000
  5900:  52.5929 028001b0ff0000
  5901:  52.5948 028001b0ff0000
  5902:  52.5968 02800100000000
  5903:  52.5988 02800100000000
  5904:  52.6008 02800100000000
  5905:  52.6028 02800100000000
  5906:  52.6048 02800100000000
[... 53 lines cut ...]
 19489:  40.6608 04fe6600000000000000000b000d001f037f007f007f7f000000000054000000
 19490:  40.6628 04fe66000000000000000009000c001d037f007f007f7f000000000054ffff00
 19491:  40.6648 04fe66000000000000000009000c001d037f007f007f7f000000000054ffff00
 19492:  40.6668 04fe66000000000000000009000c001d037f007f007f7f000000000054ffff00
 19493:  40.6688 04fe66000000000000000009000c001d037f007f007f7f000000000054ffff00
```

</details>
=> Looks like it's Hidraw 8 and 9 combined, could see the output for my button/stick inputs and the gyro when the "Z BTN" that enables mouse mode is toggled

#### Hidraw 11
=> Couldn't capture anything, idk what its purpose is, nothing logged no matter what I input.

Sidenote:
Also while retrying with sudo I saw another event to capture for DInput in evdev, so the Z BTN seems to enable a mouse mode and some buttons are transformed into mouse clicks (M3 becomes left click for example):

```
device /dev/input/event26, name "Flydigi Flydigi APEX 4", phys "usb-0000:64:00.3-1.3.2.3/input1"

Enter device path (/dev/input/event# or #): 26

Selected device `device /dev/input/event26, name "Flydigi Flydigi APEX 4", phys "usb-0000:64:00.3-1.3.2.3/input1"`.
Capabilities
{('EV_SYN', 0): [('SYN_REPORT', 0), ('SYN_CONFIG', 1), ('SYN_MT_REPORT', 2), ('?', 4)], ('EV_KEY', 1): [(['BTN_LEFT', 'BTN_MOUSE'], 272), ('BTN_RIGHT', 273), ('BTN_MIDDLE', 274), ('BTN_SIDE', 275), ('BTN_EXTRA', 276)], ('EV_REL', 2): [('REL_X', 0), ('REL_Y', 1), ('REL_WHEEL', 8), ('REL_WHEEL_HI_RES', 11)], ('EV_MSC', 4): [('MSC_SCAN', 4)]}
```

antheas, 2024-06-19:
For my convenience, what does the kernel register each hidraw as? I edited your comment to emphasize the usage pages.

yoyossef, 2024-06-19:
@antheas 

```
[ 5778.995851] hid-generic 0003:04B4:2412.002F: input,hidraw8: USB HID v1.11 Gamepad [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input0
[ 5779.003020] hid-generic 0003:04B4:2412.0030: input,hidraw9: USB HID v1.11 Mouse [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input1
[ 5779.011733] hid-generic 0003:04B4:2412.0031: hiddev98,hidraw10: USB HID v1.11 Device [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input2
[ 5779.020982] hid-generic 0003:04B4:2412.0032: hiddev99,hidraw11: USB HID v1.11 Device [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input3
```

<details>
<summary>Full dmesg output</summary>

```
[ 5777.349019] usb 7-1.3.2.3: device descriptor read/64, error -71
[ 5777.606081] usb 7-1.3.2.3: New USB device found, idVendor=04b4, idProduct=2412, bcdDevice= 1.00
[ 5777.606088] usb 7-1.3.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5777.606092] usb 7-1.3.2.3: Product: Flydigi APEX 4
[ 5777.606094] usb 7-1.3.2.3: Manufacturer: Flydigi
[ 5777.816459] usbhid 7-1.3.2.3:1.0: can't add hid device: -71
[ 5777.837778] usbhid: probe of 7-1.3.2.3:1.0 failed with error -71
[ 5777.846263] usbhid 7-1.3.2.3:1.1: can't add hid device: -71
[ 5777.859488] usbhid: probe of 7-1.3.2.3:1.1 failed with error -71
[ 5777.868440] usbhid 7-1.3.2.3:1.2: can't add hid device: -71
[ 5777.881488] usbhid: probe of 7-1.3.2.3:1.2 failed with error -71
[ 5777.890312] usbhid 7-1.3.2.3:1.3: can't add hid device: -71
[ 5777.903487] usbhid: probe of 7-1.3.2.3:1.3 failed with error -71
[ 5777.905741] usb 7-1.3.2.3: USB disconnect, device number 48
[ 5778.686468] usb 7-1.3.2.3: new full-speed USB device number 49 using xhci_hcd
[ 5778.807075] usb 7-1.3.2.3: New USB device found, idVendor=04b4, idProduct=2412, bcdDevice= 1.00
[ 5778.807087] usb 7-1.3.2.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5778.807091] usb 7-1.3.2.3: Product: Flydigi APEX 4
[ 5778.807094] usb 7-1.3.2.3: Manufacturer: Flydigi
[ 5778.995390] input: Flydigi Flydigi APEX 4 as /devices/pci0000:00/0000:00:08.3/0000:64:00.3/usb7/7-1/7-1.3/7-1.3.2/7-1.3.2.3/7-1.3.2.3:1.0/0003:04B4:2412.002F/input/input72
[ 5778.995851] hid-generic 0003:04B4:2412.002F: input,hidraw8: USB HID v1.11 Gamepad [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input0
[ 5779.002720] input: Flydigi Flydigi APEX 4 as /devices/pci0000:00/0000:00:08.3/0000:64:00.3/usb7/7-1/7-1.3/7-1.3.2/7-1.3.2.3/7-1.3.2.3:1.1/0003:04B4:2412.0030/input/input73
[ 5779.003020] hid-generic 0003:04B4:2412.0030: input,hidraw9: USB HID v1.11 Mouse [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input1
[ 5779.011733] hid-generic 0003:04B4:2412.0031: hiddev98,hidraw10: USB HID v1.11 Device [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input2
[ 5779.020982] hid-generic 0003:04B4:2412.0032: hiddev99,hidraw11: USB HID v1.11 Device [Flydigi Flydigi APEX 4] on usb-0000:64:00.3-1.3.2.3/input3
```

</details>

antheas, 2024-06-19:
Seems like `Hidraw 9: Usage Page: 0x0001 Usage: 0x0001 500hz` is a mouse for the weird thing. 

`Hidraw 8: Usage Page: 0x0001 Usage: 0x0005 500hz` is the dinput controller with all the buttons. Nice of them to include all of them. Unlike other manufacturers.

Then, I can only assume here and say that 10 is the vendor interface for the app and 11 is for firmware updates. To know more youd have to dump the hid descriptors with https://github.com/hhd-dev/hid-tools

Gyro and mouse are different though. Can you see the gyro values in one of the reports always?

antheas, 2024-06-19:
eg if hidraw 10 always outputs gyro we are in business. Otherwise, you'll have to choose between buttons and gyro

antheas, 2024-06-19:
Actually scratch that. We can just mute the mouse thing and it will be fine.

yoyossef, 2024-06-19:
@antheas hidraw 10 always outputs gyro once the mouse thing has been toggled.

It reports both the gyro (when enabled) and all the buttons (from what I've tested so far), in both mouse and gamepad mode.

antheas, 2024-06-19:
Yes. Your device is a candidate for both hhd and a kernel driver.

I would steer towards a kernel driver, for the lower CPU utilization and the fact that it can be upstreamed and such.

That driver would just black list the first 2 hidraws while active and read the one with everything 

But it would not be too hard to add support for it in hhd. Perhaps as a proof of concept for the kernel driver. Much easier than a kernel driver for sure.

The only problem right now is that hhd does not support plugin autodetection. But I had planned ahead when I made it so that is not that big of a deal.

antheas, 2024-06-19:
If the device has a windows application, open it up with wireshark and do some dumps. Theres a chance we can control gyro and the activation of other devices from it, as well as RGB if it has it. 

https://github.com/hhd-dev/hwinfo

yoyossef, 2024-06-19:
Superb news, yeah the Windows application has a lot of fancy features, the controller even has an OLED screen you can display your favorite GIF on loop, as well as LEDs.

Some of the configuration can be done on the OLED display directly, switching saved profiles for example, but lots of customization is done through the Windows app.

I'll try to do some dumps later in the evening.

zany130, 2024-06-20:
Since I also have the controller, I went ahead and did the wireshark dumps (sorry @yoyossef hope you don't mind). you can see them here https://github.com/zany130/hwinfo/tree/master/devices/Apex4/peripherals/windows

I tried to match what I saw for some other devices, but if I being honest I have no clue what I am doing, lol 

I probably should have separated some of the caps. For example, in RGB, I just switched through all the profiles and then finally off. Maybe I should have separated it by profile?

Same with the joystick dump, I just went through all the options on that page (maybe should have separated joystick mapping and joystick sensitivity)

As you can see from the names of my captures, the controller has adaptive triggers (just like the Sony DualSense) and can even emulate the DualSense to get dual sense adaptive trigger effects in games. Unfortunately, I don't have any of the supported games, so I could only capture the toggle (a.

antheas, 2024-06-20:
How do you switch to dualsense?

You can select the USB device before entering the capture through the gear 

Then while capturing you can do filters to focus in on what's going on

https://github.com/hhd-dev/hwinfo/tree/master/devices/legion_go/peripherals#introduction

yoyossef, 2024-06-20:
Dualsense mode seems to be d-input that is activated only for a few selected exes from their software (like Overwatch and a few others), and it enables Adaptive Trigger and/or gyro support.

Steam also happens to consider it as being a proper Dualsense when that mode is on it looks like, exes can be spoofed according to reddit: https://www.reddit.com/r/Controller/comments/1bez2vn/flydigi_apex_4_how_to_force_ds_mode_on_any_game/

antheas, 2024-06-20:
D input in general uses the HID protocol

The dualsense controller presents at least two interfaces: one HID device for the controller itself. Hhd can emulate that. 

And a multi channel audio device for the adaptive triggers. Hhd cannot, and there is no plans for it. However, if we can get the controller to do it that's that.

The dualsense edge controller is exactly the same as the normal variant. The only additional part is when you press the back buttons, they appear on the report. 

 If that's the case, there's potential for adding support for back buttons in steam together with gyro without needing any driver. And getting adaptive triggers as a bonus.

They only activate when the controller uses the edge vid pid in sdl and steam, so you won't see them right now. But they will be in the report.

antheas, 2024-06-20:
Sorry I mispoke its morning 

Adaptive triggers are part of the hid report

The audio is for the fancy rumble

yoyossef, 2024-06-20:
No worries, the difference with the dualsense is the lack of trackpad and lack of fancy haptic feedback (in fact I've heard that in d-input the vibration motors are completely off, I'd have to verify this), so we would have to capture and understand the events when that special Dualsense mode is on.

I can do it over the week-end unless zany does/did it.

antheas, 2024-06-20:
The dualsense protocol is well documented

You just need to capture whether the back buttons work in that mode. Probably not. And if the audio devices exist

Doing rumble in dualsense mode is a bit complex but quite standard. Including rgb.

But maybe they skipped that whole receiving  part and just do some emulation for the triggers. Since you can still see the original controller

antheas, 2024-06-20:
As for a bit of context on rumble: while the dualsense audio device is active the dualsense is replaying the audio as rumble

In parallel, the games that send audio to the controller also send a compatibility rumble in the configuration report that is classic rumble. The dualsense is supposed to ignore this

Unless the disable audio effects bit is set. Steam, proton, the kernel and games that do not support the special effects, turn that but on. So genuine controllers always have rumble

We had to figure that out to get rumble working in death stranding that uses the audio rumble and has that bit off

The led effects on that are also quite special and they work under hhd. Unfortunately not for GPD

zany130, 2024-06-20:
> How do you switch to dualsense?
Unfortunately, you can only do this in the software 

see https://github.com/zany130/hwinfo/blob/master/devices/Apex4/peripherals/windows/adaptive_trigger_milesmorales.png

> Then while capturing you can do filters to focus in on what's going on
> 
> [hhd-dev/hwinfo@`master`/devices/legion_go/peripherals#introduction](https://github.com/hhd-dev/hwinfo/tree/master/devices/legion_go/peripherals?rgh-link-date=2024-06-20T05%3A49%3A00Z#introduction)

Oh, I didn't see that.

So, should I redo the dumps with filters?

antheas, 2024-06-20:
Filters do not affect the dump. Selecting the USB device does. 

If the files are not too big, no. If they are yes. Also, some sensitive info might have snuck through.

yoyossef, 2024-06-20:
https://github.com/zany130/hwinfo/pull/1

I've captured some traffic in said "DS Mode" by spoofing the Overwatch 2 exe with Chrome (like in the reddit example I had linked above).

zany130, 2024-06-20:
> [zany130/hwinfo#1](https://github.com/zany130/hwinfo/pull/1)
> 
> I've captured some traffic in said "DS Mode" by spoofing the Overwatch 2 exe with Chrome (like in the reddit example I had linked above).

Nice thanks. Does overwatch have "standard" or "DS mode" adaptive triggers ? I'm guessing since you said "DS Mode" it has that. I ask because I remember a lot of the games on the adaptive trigger list only having "standard" adaptive triggers

> Filters do not affect the dump. Selecting the USB device does.
> 
> If the files are not too big, no. If they are yes. Also, some sensitive info might have snuck through.

ok then my dumps should be fine the biggest one was like 10mb  and most where under 5mb

yoyossef, 2024-06-20:
I've also [captured traffic](https://github.com/yoyossef/hwinfo/commit/90ce105f1ae99c04a4b6e357ee20705acf4b302c) for the GIF upload thing (sidenote: the screen is always on so having an option to upload a black GIF to turn it off could be a nice feature to have).

But for some reason the flydigi app requires to switch to Xinput, I've captured it anyway, I won't PR it yet as it's quite a big file, I should filter it once I can reliably identify which traces are what.

> Does overwatch have "standard" or "DS mode" adaptive triggers ?

Overwatch only offers DS mode.

zany130, 2024-06-20:
> I've also [captured traffic](https://github.com/yoyossef/hwinfo/commit/90ce105f1ae99c04a4b6e357ee20705acf4b302c) for the GIF upload thing (sidenote: the screen is always on with the GIF looping so having an option to upload a black GIF to turn it off could be a nice feature to have).
> 
> But for some reason the flydigi app requires to switch to Xinput, I've captured it anyway, I won't PR it yet as it's quite a big file, I should filter it once I can reliably identify which traces are what.

Yeah a lot of the features of the app require the controller to be in xinput mode. I should mention all my captures where done in xinput mode.

the reason being it will sometimes randomly switch you to xinput mode if your on dinput

yoyossef, 2024-06-20:
The app does reset itself and sets the pad to xinput quite randomly but I wouldn't say that xinput is required to change those settings necessarily.

My guess is that we should capture as much dinput traffic as possible, what do you think @antheas ?

(Also curious if you're seeing interesting stuff in my DS mode captures)

antheas, 2024-06-20:
Xinput is probably required because as you said the app is unstable. So for the app to detect the controller, the controller has to be in xinput mode. No need for them to test all the modes with it

antheas, 2024-06-20:
> My guess is that we should capture as much dinput traffic as possible, what do you think @antheas ?

My next move would be to start finding commands and move over to linux. Try to find the commands that make it change modes, then you can run them with `hhd.controller.lib.hid` and make it switch modes in linux and study it there.

zany130, 2024-06-27:
found this https://github.com/ahungry/vader3/tree/master 

While not the Apex 4 controller, it is still a Flydigi controller. If you're still looking at this, @yoyossef, it may be useful?

antheas, 2024-06-27:
you can add support to steam for back buttons and gyro if the driver picks up the buttons

antheas, 2024-06-27:
Only dualsense and switch controllers support gyro, so youll get one of those glyphs

At least until this releases:
https://hori.jp/products/hpc/hpc-055/

zany130, 2024-06-28:
Interesting updates: apparently, Gyro sort of works in xinput mode on Linux.

when configured like this 

[image]

I have the power-on button set to LB (no idea why its showing blank). This actually works on Linux (ie holding LB moves the left joystick according to the gyro)

I wonder if there is some way to grab the gyro output, ignore the joystick movement, and then use that as a gyro in Steam.

since this is xinput mode vibration also works (vibration does not work in dinput even in Windows with the software running, nor does it work in "DS mode")

zany130, 2024-06-29:
Added some more hid report dumps into https://github.com/zany130/hwinfo/tree/master/devices/Apex4/peripherals/hid

sorry if some of this is redundant, as @yoyossef  already did most of this, but ...

interestingly, I found that the 3rd hidraw device in dinput mode actually does outputs the gyro without the mouse when you have the mouse toggle off (circle button next to home)

so the hidraw devices are as follows

1st. All controller inputs plus mouse gyro and buttons when mouse mode is toggled (circle button)
2nd. only mouse gyro and buttons (when mouse mode is on)
3rd. All controller inputs plus gyro when mouse mode is off. When on, gyro mouse and mouse buttons
4th.  firmware updates or something?

like I said above, I didn't notice any difference in the hidraw reports, between normal dinput mode  and getting the controller to go into "DS mode" through a Windows virtual machine

zany130, 2024-06-30:
Found a really nice app to test the controller on windows https://powgames.itch.io/game-controller-tester

and found out some interesting things

1. in dualsense mode, the controller is advertised as having all the buttons of a dualsense and no more. so no back buttons, but the original input controller is still connected and that does still send inputs from the back buttons so I guess both devices could be combined? All dualsense buttons work except the mute button (touch pad press is mapped to Circle button next to home no touchpad swipes)
2. There is no vibration in dinput or dualsense mode. Vibration support is advertised in dualsense mode ,but does not work
3. As I expected, dinput mode does have an accelerometer and gyro sensors! steam just doesn't recognize it from some reason on linux or even on windows

so in summary it should be possible to have dinput mode expose both gyro and accelerometer sensors aswell as the extra back buttons and the extra circle button to steam, but we may not be able to get rumble working

zany130, 2024-06-30:
got steam to recognize the extra buttons with this env

`SDL_GAMECONTROLLERCONFIG="0300fe68b40400001224000011010000,Flydigi APEX 4,a:b0,b:b1,x:b3,y:b4,back:b10,guide:a4start:b11,leftstick:b12,rightstick:b13,leftshoulder:b6,rightshoulder:b7,dpup:h0.1,dpdown:h0.4,dpleft:
h0.8,dpright:h0.2,leftx:a0,lefty:a1,misc1:b15,paddle1:b14,paddle2:b16,paddle3:b17,paddle4:b18,rightx:a2,righty:a3,lefttrigger:b8,righttrigger:b9,platform:Linux,`

I switched `guide` to `a4` and `misc1`  to `b15` to have the guide button be the circle button and the misc button be home. this works better, I think, because holding home triggers the on-screen menu, making it difficult to perform steam chorded inputs like `guide` + `a` for the quick-access menu

no idea how to hide the gyro mouse on circle button or how to map the gyro though...

antheas, 2024-07-01:
Except vibration is a strong statement. Maybe the driver does not know how to make the controller vibrate

But from what you're saying this is a kernel driver candidate 

Mind the fact that the only kernel drivers that support gyro are the steam deck, playstation and Nintendo ones

So you'll need to do some copying around

zany130, 2024-07-01:
> Except vibration is a strong statement. Maybe the driver does not know how to make the controller vibrate

I was basing the lack of vibration more on the fact that even on Windows, there is no vibration when in either dinput mode or "DS mode" (which I think is just dinput mode with the product and vendor ID of the dualsense?).

Is the best path forward to copy Windows software "DS mode" and bind the controller to the kernel PlayStation hid driver?

That way we may also be able to get per-game adaptive triggers

Unfortunately, this is beyond my ability. I just know how to do basic tinkering lol :sweat_smile: 

but it does look like what prevents steam from "seeing" the gyro in dinput mode is lack of gyro support by usbhid driver? (I think that's what get used)

Stupid question but would it be possible to force it to use the PlayStation hid driver?

antheas, 2024-07-01:
Adaptive triggers are not part of the playstation driver. Linux has poor support for adaptive triggers so I don't think it's very worthwhile. Also back buttons which are more important would be missed.

The best plan forward is to first document the dinput protocol, and then either make a patch to SDL or the kernel.

Steam uses the SDL library which is maintained by valve for controller support. The SDL library supports the kernel drivers for gyro and all buttons, if they are exposed as the Nintendo and Playstation drivers expose them.

SDL also has a large number of hand built drivers that connect to controllers directly through hid. And through that it supports touchpads, gyro, and RGB for most Microsoft, Nintendo, and Sony controllers.

As far as steam is concerned, SDL contains a set of functions that receive a pid and vid and return if the controller is a Dualsense Edge controller, Dualsense, Nintendo, etc.

Steam input calls those functions and depending on which one returns yes activates a different profile.

Then, provided that Sdl has loaded the controller properly, those features work. However, you can have cases where say SDL cannot access the raw hid device. In that case, it falls back to the kernel driver which has a subset of features. So for example, a dualsense edge controller would still show up as if it has back button and RGB support but those features would not work.

antheas, 2024-07-01:
Which is a roundabout way of saying that if you want back buttons without valve's blessing, your controller has to identify as Xbox elite, Dualsense edge, or a joycon pair

And if you want gyro, it needs to identify as a Dualsense edge or Nintendo controller. It is not possible to get steam input to support gyro without weird glyphs.

You will also be able to see that when gyro or RGB works, they will be in the advanced controller settings in Steam. But if there's not a case for them in SDL Steam Input will not let you use them

zany130, 2024-07-02:
Again, I may be completely out of my depth here and wrong, but wouldn't it be possible for some userspace driver application like Xboxdrv or SC-controller (and I am assuming HD works similarly) to grab the Hidraw device that has gyro and everything and expose it as a controller, preferably a dual-sense controller?

The problem is that I think Xboxdrv and sc-controller don't support Gyro. 
The Vader 3 driver I found doesn't seem to support Gyro either (though  I think the Vader 3 also has Gyro?)

I know the ideal solution would be to deal with the controller upstream, but this could be POC or something.

I already got the back buttons sort of working on Steam with the SDL mapping. However, as you said, since it's not identified as a DualSense Edge or any of the other known controllers that have paddle support, it doesn't activate the right profile for Steam input, meaning. At the same time, they are detected in the controller tester however, you can't map them.

though interestingly I found if you apply a steam input profile from a known supported controller with paddles (like the deck or dual sense edge) you can get buttons mapped to the back paddles

Just tried this with Persona 3 reload and I was able to get some actions mapped to the back buttons

antheas, 2024-07-02:
Oh didn't know about that workaround or I guess steam bug. Yes of course you can even remap the dinput controller into a dualsense edge with hhd.

Hhd has support for dinput, xinput, and hidraw. If dinput works properly use that.

Hidraw at 1000hz would use around 10% of a single core. If you do dinput that drops to 0 unless you're using the sticks. In which case you're looking at 7-10% again. Only while you're moving the sticks.

Id look for a way of spoofing the vid/pid of the controller so steam thinks it's a dualsense edge out of the gate and you don't eat the CPU cost.

Reminder we only do this hack for handheld devices because certain buttons come from a separate device and it's not possible to merge them in the kernel.

Also, open an issue on SDL and reference this issue. I'm sure they would be happy to help and maybe steam gets support for this controller.

antheas, 2024-07-02:
I also benchmarked hhd yesterday and you would be happy to know it only adds .1ms to .5ms of latency depending on how the kernel feels about waking up the threads.

zany130, 2024-07-02:
> Also, open an issue on SDL and reference this issue. I'm sure they would be happy to help and maybe steam gets support for this controller.

I will do that. Ideally, it gets fixed there. Honestly, it sounds like the same amount of effort would need to be made to add it here or Steam/SDL. Sorry if I was bugging you or anything, but like I said, I've actually had zero experience with all this.

Thanks for pointing me and @yoyossef in the right direction!

antheas, 2024-07-02:
No problem guys, getting better controller support is a win win for everyone.

zany130, 2024-09-26:
Just to give an update, it seems that with what we know of the controller right now, it won't be feasible to add support in SDL.

I am still considering documenting the controller when I have time. I have just been really busy lately.

antheas, 2025-06-30:
From what I have seen, more controllers have steaminput support. This is semi out of scope now so closing.

closed 2025-06-30 by antheas