hhd-dev/hhd #74: Per Profile TDP not working on GPD Win Mini 2024 (8840u)

TheEtherNetBoyz, 2024-07-01:
Per Profile TDP is not working on the 2024 Win Mini, overall TDP is working from HHD plugin, but not per profile via steam TDP slider. Not sure if its due to the 8840u or not.

antheas, 2024-07-01:
Try to reboot. Steam need to restart for the slider to activate. If that still does not fix it, we can go more indepth.

TheEtherNetBoyz, 2024-07-01:
Just tried, its still reading the HHD slider and not the per game profile

antheas, 2024-07-01:
So the steam slider works

Does the steam slider control the hhd slider?

TheEtherNetBoyz, 2024-07-01:
No they're acting as two separate things. the HHD slider works for TDP and only changes when i adjust it in HHD. The per game profiles with steam do nothing.

antheas, 2024-07-01:
That's a bug. The steam slider should control Handheld daemon in real time

Just not the other way around.

TheEtherNetBoyz, 2024-07-01:
If you need me to test anything lmk

antheas, 2024-07-01:
Fixed. Your other issue with the button also got fixed but is not released yet.

https://github.com/hhd-dev/adjustor/releases/tag/v3.0.1
https://github.com/hhd-dev/adjustor/commit/10b3576e253d128c9f883ab97f4cbc5b6a36bd1f

closed 2024-07-01 by antheas