hhd-dev/hhd #75: TDP slider out of sync (ROG Ally)

codevski, 2024-07-04:
Ever since the latest update fix with bazzite the tdp slider seems to be all out of sync.

Handheld daemon is not in sync with steams TDP limit.

Daemon: 3.1.0
Daemon UI 3.0.1
TDP 3.0.1

antheas, 2024-07-04:
First, grab a log with`journalctl -b | fpaste` so we can have it

Then reboot. The tdp feature is a bit experimental and there are edge cases where it can stop working. I fixed most of them to the point I only managed to reproduce it once

If we can reproduce the breakage I can work on it

Unfortunately due to how it works, restarting hhd will not fix anything, you actually need to reboot

codevski, 2024-07-04:
https://paste.centos.org/view/565bfda8
https://paste.centos.org/view/d9e477b9

Mmmm tried rebooting doesn't seem to change anything. To replicate for me in steam set per game and have default to 8w, open game make sure per game is set and try set to 14, when you close TDP overlay it seems to work but soon as you open it again it goes back to 8

I'll try get video up

antheas, 2024-07-04:
First, lets keep an eye on this:
` asusd::ctrl_platform] ThrottlePolicy setting EPP` asusd daemon is interfering. @KyleGospo Another reason for removing the ally image

Here is the relevant part from the log:
```
Jul 04 23:46:12 fedora hhd[4284]: [23:46] OVRL  INFO     Executing overlay command: 'open_qam'
Jul 04 23:46:12 fedora hhd[4284]:         OVRL  INFO     UI: Switching ui to 'qam'
Jul 04 23:46:12 fedora hhd[4284]:         ADJS  INFO     Received TDP value 14 from /sys.
Jul 04 23:46:13 fedora hhd[4284]:         ADJS  INFO     Received TDP value 8 from /sys.
Jul 04 23:46:15 fedora hhd[4284]:         ADJA  INFO     Setting platform profile to 'quiet'
Jul 04 23:46:15 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:15 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:15 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:15 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'fast' to 8 by writing to:
Jul 04 23:46:15 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_fppt
Jul 04 23:46:15 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'slow' to 8 by writing to:
Jul 04 23:46:15 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_pl2_sppt
Jul 04 23:46:15 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'steady' to 8 by writing to:
Jul 04 23:46:15 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_pl1_spl
Jul 04 23:46:16 fedora hhd[4284]:         ADJS  INFO     Received TDP value 8 from /sys.
Jul 04 23:46:16 fedora hhd[4284]:         ADJS  INFO     Received TDP value 14 from /sys.
Jul 04 23:46:18 fedora hhd[4284]:         ADJA  INFO     Setting platform profile to 'balanced'
Jul 04 23:46:19 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:19 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:19 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:19 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'fast' to 14 by writing to:
Jul 04 23:46:19 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_fppt
Jul 04 23:46:19 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'slow' to 14 by writing to:
Jul 04 23:46:19 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_pl2_sppt
Jul 04 23:46:19 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'steady' to 14 by writing to:
Jul 04 23:46:19 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_pl1_spl
Jul 04 23:46:19 fedora hhd[4284]:         OVRL  INFO     Executing overlay command: 'open_qam'
Jul 04 23:46:19 fedora hhd[4284]:         OVRL  INFO     UI: Switching ui to 'qam'
Jul 04 23:46:20 fedora hhd[4284]:         ADJS  INFO     Received TDP value 14 from /sys.
Jul 04 23:46:20 fedora hhd[4284]:         ADJS  INFO     Received TDP value 8 from /sys.
Jul 04 23:46:21 fedora hhd[4284]:         OVRL  INFO     Executing overlay command: 'open_qam'
Jul 04 23:46:21 fedora hhd[4284]:         OVRL  INFO     UI: QAM is currently open, closing.
Jul 04 23:46:21 fedora hhd[4284]:         OVRL  INFO     UI: Switching ui to 'closed'
Jul 04 23:46:21 fedora hhd[4284]:         ADJS  INFO     Received TDP value 8 from /sys.
Jul 04 23:46:22 fedora hhd[4284]:         ADJS  INFO     Received TDP value 14 from /sys.
Jul 04 23:46:24 fedora hhd[4284]:         ADJA  INFO     Setting platform profile to 'balanced'
Jul 04 23:46:24 fedora asusd[4696]: [INFO  asusd::ctrl_platform] ThrottlePolicy setting EPP
Jul 04 23:46:24 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'fast' to 14 by writing to:
Jul 04 23:46:24 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_fppt
Jul 04 23:46:24 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'slow' to 14 by writing to:
Jul 04 23:46:24 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_pl2_sppt
Jul 04 23:46:24 fedora hhd[4284]:         ADJA  INFO     Setting tdp value 'steady' to 14 by writing to:
Jul 04 23:46:24 fedora hhd[4284]:                        /sys/devices/platform/asus-nb-wmi/ppt_pl1_spl
```

Seems like it keeps receiving 8w and 14w from Steam. And it is working correctly.

Are you on the steam beta?

antheas, 2024-07-04:
Oh i think i might know why that is, when the overlay opens it thinks Handheld Daemon is a game? That's new

codevski, 2024-07-04:
Nope using stable not on beta:

https://youtu.be/ybmraoi4Ou8

antheas, 2024-07-04:
Yeah... it thinks hhd is a game... I will fix this for the next version...

Does closing the overlay fix the tdp after a bit?

codevski, 2024-07-04:
yeah it kinda goes back to what it is set on steam as you can see in the vid fps goes back up. But setting tdp level in the overlay doesnt change the value in the steam slider. For exaple if your just in steam not in game and you set tdp in overlay to 9, the steam slider will stay on 8 (also the slider in decky handheld daemon is still on 8) but the overlay is still holding the 9w value lol

antheas, 2024-07-04:
Yes, I cant change what steam does unfortunately, can only read from it. I just do what it says.

So the solution to this will be to make it so hhd pretends to be the game as well

codevski, 2024-07-04:
Hrmm how would that work with per game profiles, if you've set per game and then adjust on on the overlay would you need to manually set it within steam and just use the overlay as a read-only / temp set TDP until you launch the game again?

antheas, 2024-07-04:
If you are in a game where you have enabled the Steam TDP slider, Steam will periodically update the TDP so you need to use steam for that.

However, if you disable the slider for a certain game (as in no TDP limit) or you are in desktop you can use the overlay/app to set TDP.

There's no conflict between using both sliders, but Steam will always override the hhd slider. This is the best I can do unfortunately.

As for the bug you reported in this thread, I will attempt to fix it the following days. Seems like just pretending to be steam confuses steam and makes it blank, so the solution will have to be more elaborate.

codevski, 2024-07-04:
Yeah fair enough, unfortunately having TDP limit off per game will mean that the TDP set will be the same in and out of games where I kinda need different TDP limits per game for the sake of getting as much battery

antheas, 2024-07-04:
So you either need to use tdp limit on all games or none?

Unfortunate. Guess the solution to this will be making the hhd slider read only while Steam controls the tdp

Or that I should finally add per game profiles

codevski, 2024-07-04:
Yeah need I tend to need to use on all games to min max battery usage, per game profiles would be awesome and to fully replace simpledecky would be two profiles per game, (AC / battery) without the need to constantly changing tdp

antheas, 2024-07-06:
Hopefully this fixes it: https://github.com/hhd-dev/hhd/commit/cb399dcd785e4643cd0991d69016ea8339adfb43

It did for me after testing. Steam will no longer get confused with per-game profiles

Use `ujust _hhd-dev` on bazzite to test it out as well.

closed 2024-07-06 by antheas