hhd-dev/hhd #77: Minecraft Bedrock Edition NIX Launcher Constant Scrolling

davey1343, 2024-07-07:
Device is ROG ALLY.

When using the Bedrock launcher on Bazzite it is not possible to play the game as the HHD input is constantly being read causing continuous scrolling downwards.

antheas, 2024-07-07:
Can you link the launcher?

davey1343, 2024-07-07:
https://github.com/minecraft-linux/mcpelauncher-ui-manifest/tree/ng

davey1343, 2024-07-07:
> It's still occuring. It only occurs in game mode. Running from desktop is
> fine.

This is after running the command:

https://photos.app.goo.gl/hNDsUz5qHYAQGJFX7

If I hold command centre to go into mouse mode and move cursor it will stop
but then start again if I move the left stick:
https://photos.app.goo.gl/nyij64otrS8zEPoK6

antheas, 2024-07-07:
Before you launch the game go into controller settings and disable steam input

See if it still does it

davey1343, 2024-07-07:
That has stopped it but obviously have no gamepad controls now

antheas, 2024-07-07:
You should still have controllers work. Steam input only does remapping. If the launcher does not support controllers that's another problem and you need to deal with your steam input profile

davey1343, 2024-07-07:
Ok.

Steam input off and then rebinding the controls within the launcher and the
issue reappears 😭

davey1343, 2024-07-07:
I have another input called ' ASUS Keyboard' within the launcher button
binding screen. Could that be the issue?

I'm assuming it's something to do with HHD as my son's steam deck doesn't
have the issue.

antheas, 2024-07-07:
I've had a similar issue on some games after suspend. They get stuck on left stick in the menus. Haven't figured out why

Probably not the Asus keyboard although it wouldnt be the first time it interfered

antheas, 2024-07-07:
Do you have a more reputable link for the launcher? I can download it later/tomorrow

Would also be nice to demo it on the steam deck

It's hard to figure out how things should behave when I don't have a steam deck

davey1343, 2024-07-07:
Ah ok. I'm not mad then. May be useful for solving your resume issue then
as it's a repeatable issue.

davey1343, 2024-07-07:
It's available on the discover launcher within bazzite

davey1343, 2024-07-07:
This is on my son's steamdeck:

https://photos.app.goo.gl/AsACkTwric7iw9z89

antheas, 2024-07-07:
https://flathub.org/apps/search?q=minecraft should be in here somewhere?

davey1343, 2024-07-07:
Minecraft Bedrock Launcher. 2nd one down

antheas, 2024-07-07:
Do you need an android license for it?

davey1343, 2024-07-07:
Ah yes you need Minecraft apk on your Google play account

antheas, 2024-07-08:
I will look into it these days. I know some other games have a similar issue. For now, I would suggest you try Steam Input disabled with the Xbox emulation profile. I cant promise you a solution this week, so try to find a workaround for now.

if the command you ran worked and you had the issue as well, then its not hhd and its an obscure steam issue. Nevertheless, it should be fixed.

antheas, 2024-07-27:
I tried various fixes for it and encountered a similar issue in different ways. We are unsure of why it happens. 

We also have had it happen with external controllers without Handheld Daemon running. It is very weird.

antheas, 2024-10-26:
Perhaps this fixed itself. We have done a lot of work on the matter and input management is a lot more mature now.

closed 2024-10-26 by antheas