hhd-dev/hhd #80: Feature Request: Disable Touchscreen

NexGen-3D-Printing, 2024-07-17:
Hi All,

Very happy with this tool, makes the Legion Go actually usable, so I really appreciate the hard work.

I would like to ask if its possible on future releases, the ability to add a button to disable the touchscreen? I find the touchscreen can really mess up the gaming experience, this happens on the Steam Deck, but is far far worse on the Legion Go, the right thumb stick is very close to the screen, I find I constantly touch the screen during gameplay and this will throw it into mouse mode, then back to controller mode constantly.

No current alternative fixes will work, Steam should really add this to their quick settings menu, but for 2 yrs, they have not done it, many have put in the request, so I think HHD could be the answer.

Once again, thanks for the hard work on this tool.

antheas, 2024-07-18:
Sure. That is a possibility. I can look into that next week.

If you just completely want to disable the touchscreen, I can make a little script for you for now

NexGen-3D-Printing, 2024-07-18:
Hi antheas, thank you for responding, 

The script wont be necessary, I can wait, but I'd be happy to test out a new version if its implemented, from what I have read on the net, I think many people will be happy with this feature, I cant believe Steam have not implemented it themselves, so many complaints on the Steam forums.

antheas, 2024-07-18:
I will be implementing screen gestures next week, so it's trivial to add it as part of that

Just a bit concerned with people forgetting it on and then saying their screen is broken

NexGen-3D-Printing, 2024-07-18:
Its bound to happen, maybe have it turn off when one exits a game or reboots the device, its easy to enable again in game from the HHD menu.

Honestly, on these handhelds, I personally never use the touchscreen, game mode works best with a controller, and the desktop mode, well Linux is great at a lot of things, but a table PC interface its not.

antheas, 2024-07-25:
Released as of 3.3.0 https://github.com/hhd-dev/hhd/releases/tag/v3.3.0

closed 2024-07-25 by antheas