hhd-dev/hhd #81: Ayaneo Kun support

darrencocco, 2024-07-20:
I've got an Ayaneo Kun and I'm willing to help with testing etc.

antheas, 2024-07-20:
Hi. of course.

I will assume the speakers do not work. I know that much.

As far as the controllers go, we will need to do some reverse engineering for that.

`hhd.contrib evdev` will let you see which devices exist and what events they emit.

You should be able to see some MSC events for the front ayaneo buttons.

For the back buttons, I need windows ayaneo space photographs, and you will probably need to use rweverything to see how they are programmed. We need to see if we can isolate them somehow.

For the touchpads, same as above.

darrencocco, 2024-07-22:
Okay, looks like the face buttons on the right hand side and the additional two shoulder buttons on the top are a "keyboard" mostly outputting some F-Keys. vis-a-vis the touch pads. For me they seem to be acting like a mouse with the left and right push being left and right click.

darrencocco, 2024-07-23:
So I've worked how the back paddle buttons work. They seem to be configured by the AYA Space program to emit certain key presses.

antheas, 2024-07-23:
Can you document those options with pictures and post them?

antheas, 2024-07-23:
Also, with hhd.contrib evdev check if they emit MSC codes

darrencocco, 2024-07-23:
> Also, with hhd.contrib evdev check if they emit MSC codes

Checked in hhd.contrib evdev and they emit the pressed and released events for the buttons I configured them for (BTN_A and BTN_B). They're part of the existing xpad controller.

I think there's a good chance they're configured through I2C.

darrencocco, 2024-07-23:
> Can you document those options with pictures and post them?

Sure bit it will take some time. I will go through and methodically capture everything in the controller configuration.

antheas, 2024-07-23:
I do not think its I2C. In my opinion, it is either USB or direct memory writes to the EC.

If it is USB, you can follow below as you do that and capture the packets:
https://github.com/hhd-dev/hwinfo/tree/master/devices#getting-the-raw-hid-descriptors

It is with wireshark and USBPcap, very safe and easy.

If it is direct memory writes to the EC, you will need RWAnything, which is a software that can read the EC memory map and to write down the memory addresses that are written.

Beware that it is a bit of a pain and shaddy to setup though.

darrencocco, 2024-07-24:
I mention I2C as the audio problem is because there's an internal amplifier that needs to be initialised by I2C.

Not a problem if it is memory writes to the EC. I can work on writing the kernel driver.

antheas, 2024-07-24:
Yes, audio amplifiers usually are connected through the i2c bus.

If it was through i2c I want to say they would have used hid and you would be able to see the device with `hhd.contrib hidraw`

darrencocco, 2024-07-26:
Attached are the dmi data and recordings for the touch-pads/he PS2 Keyboard that the extra buttons on the face/shoulders show up under. The touch-pads are both reconfigurable FYI.
[image]
[keyboard.txt](https://github.com/user-attachments/files/16387162/keyboard.txt)
[touchpad1.txt](https://github.com/user-attachments/files/16387164/touchpad1.txt)
[touchpad2.txt](https://github.com/user-attachments/files/16387165/touchpad2.txt)
[dmi.txt](https://github.com/user-attachments/files/16387175/dmi.txt)

antheas, 2024-07-26:
Thanks for the data. Yes the keyboard is standard in Ayaneo devices, but seems not completely in this case.

## Touchpads
I looked at the touchpad information. It is not a raw touchpad. This means Ayaneo is finally doing custom USB stuff. The touchpads are connected to the Kun MCU, which remaps them. This means that we need to first configure it in a standard way, then layer our customizations on top.

```
 - device /dev/input/event4, name "AYANEO AYANEO COMPOSITE DEVICE", phys "usb-0000:c4:00.3-4.2/input0"
 - device /dev/input/event5, name "AYANEO AYANEO COMPOSITE DEVICE", phys "usb-0000:c4:00.3-4.2/input1"
 ```

In this case, it is also very likely that the touchpads and paddles are configured through USB. Therefore, we need the HID descriptors of the composite device.

Therefore, you should use [hid-tools](https://github.com/hhd-dev/hid-tools) to dump the descriptors from all the hidraw devices that appear to come out of this interface. The git has instructions there.

Just cut out the descriptor from the paste. We do not need the events.

This will tell us if the device has vendor hid reports that are used to configure the device. Of course, being a new manufacturer, Ayaneo might have gimped the descriptors and even if there are configuration commands they might be missing.

Then, you can use Wireshark with USBPcap in windows to sniff the configuration data from the device. Given the hid descriptors, fishing them out will be easy.

## Keyboard
HHD already contains a systemd rule about Ayaneo devices that maps their scancodes to keycodes. But it seems like the Kun has a different set of scan codes. Therefore, I upgraded `hhd.contrib evdev` to print the scan codes in hex.

Can you redo the keyboard recording with the new version from git, and put a name to the scancodes?

Preview:
```
┌─────────────────────────────────────────────────────────────────┐
│ 103.708s / EV_MSC (004) /              MSC_SCAN (004):   0x00B0 │
│ 103.708s / EV_KEY (001) /          KEY_VOLUMEUP (115): released │
└ SYN ─ 104.809s ─    6.5 Hz ─────────────────────────────────────┘
```

darrencocco, 2024-08-05:
[image]
[hidraw0_recording.txt](https://github.com/user-attachments/files/16494053/hidraw0_recording.txt)
[hidraw1_recording.txt](https://github.com/user-attachments/files/16494054/hidraw1_recording.txt)
[hidraw2_recording.txt](https://github.com/user-attachments/files/16494055/hidraw2_recording.txt)
[scancodes_LC.txt](https://github.com/user-attachments/files/16494057/scancodes_LC.txt)
[scancodes_RC.txt](https://github.com/user-attachments/files/16494058/scancodes_RC.txt)
[scancodes_TouchButton.txt](https://github.com/user-attachments/files/16494059/scancodes_TouchButton.txt)
[scancodes_AYAButton.txt](https://github.com/user-attachments/files/16494060/scancodes_AYAButton.txt)
[scancodes_=Button.txt](https://github.com/user-attachments/files/16494061/scancodes_.Button.txt)

antheas, 2024-08-05:
hidraw2 is a configuration interface. You should be able to sniff it with Wireshark and USBPcap in windows.

Just select the USB device of the controller with the gear icon before recording.

And make sure to launch it with administrator priviledges. So it does not open UAC between recordings.