hhd-dev/hhd #84: THIS UPDATE BROKE HHD

Evilexium, 2024-07-31:
Well everything worked fine on CachyOS Rog Ally, Since 1 or 2 days it makes strange issues since HHD Latest Updates. If i go sleep mode the LEDS act weird low dimmed red and wake up the ally no RGB no Controller support have to Hard reset the device everytime. 

Indeed not very nice since 2 updates before everything worked as indeed , i feel like all this Fast Out off control Updates u guys make with HHD , just ruins this good software into Oblivion. 

There are problems with RGBs , Spiral doesnt do anything beside stay Black , Sleep Mode break the controller support and HHD RGB LEDS , and controll of UI. The Configurations ask for token often even if the token is right it doesnt connect it say DOES HHD RUNN ?

Its annoying as F*** at this moment :)

antheas, 2024-07-31:
Unfortunately, this is a kernel problem related to cachyos. I would direct you to its developers

antheas, 2024-07-31:
I made a report on your behalf

antheas, 2024-07-31:
I have reported the issue to cachyos and they are aware of it. I will keep this issue open until it is resolved.

Evilexium, 2024-07-31:
Thank you Sir :) hope that video help them identify whats wrong since i didnt know how to explain it better .

antheas, 2024-08-05:
Seems they have made some changes that I think might not have helped. Do you still have the issue after updating?

Evilexium, 2024-08-05:
I did yesterday fresh install still was there but with a annyoing patern , 4 times in row no LED after wakeup but controlls in game worked , after that 3 times in row Led worked and controlls only overlay worked but the in game controlls didnt . Today i did fresh instal again like yesterday and now seems to work dont know how long but will write here how it runs. I dont see big changes since yesterday its same kernel 6.10.3 but still maybe they did some tweaks :) . Thank you for all this effort sir

philmmanjaro, 2024-08-19:
@Evilexium **CachyOS** is currently not compatible unless **hhd** gets updated to upstream changes, which CachyOS kernel already adopted: https://github.com/CachyOS/linux/commits/6.10/handheld/. It is recommended to use [InputPlumber](https://github.com/ShadowBlip/InputPlumber/releases) plus [OpenGamepadUI](https://github.com/ShadowBlip/OpenGamepadUI/releases) for TDP control. We will see when **hhd** will catch up on those changes. More in depth discussion on the matter here: https://lore.kernel.org/lkml/202408071743.00IxSKrf-lkp@intel.com/T/

antheas, 2024-08-19:
Hi Philip

First off, the patch you sent is not upstreamed yet, including having multiple compile warnings and bugs such as random vibration 

You can read my notes in lkml

Secondly, it does not affect the Ally. Another patch that was sent by Mr Jones in a parallel thread was broken, which caused the ally to break. It is my understanding that Mr Jones has released an updated version of this patch that CachyOS uses now

Evilexium, 2024-08-19:
> @Evilexium **CachyOS** is currently not compatible unless **hhd** gets updated to upstream changes, which CachyOS kernel already adopted: https://github.com/CachyOS/linux/commits/6.10/handheld/. It [...]

It works perfectly now , I have CachyOS Handheld + HHD + Decky Loader + SimpleDeckyTDP + RyzenADJ = everything works , Rainbow Spiral RGB , Suspend/ Sleep Mode while in game , Controller , and TDP controll from SDTDP. No more issues.

No need for InputPlumber and OpenGamepadUI

antheas, 2024-08-22:
Let us assume this matter is resolved, as I have not seen anyone complaining about it in a while.

Evilexium, 2024-09-05:
Yeah Wellcome back DAMN ISSUE is here again since last Update either from CachyOS kernel or from HHD . Dont know but its 1 to 1 the same . everything worked perfectly almost a month now its back this MF ISSUE :(