hhd-dev/hhd #85: Parts of install script fail

kspes, 2024-08-05:
Hi,

I've successfully installed hhd on legion go (debian trixie/sid), but in the process of doing so, I've noticed that the install script printed several errors which resulted in a semi-sucessfull installation.

To fix all of them, I had to install the following packages in debian:

- libgirepository1.0-dev
- libcairo2-dev
- libpython3-dev
- python3-venv
- libhidapi-hidraw0

However, what would've been more informative is if the script failed at the first error instead of continuing.
This can be simply achieved in a bash script  by adding teh line "set -e" at the top of the file. This basically means the script will stop executing on the first error.

Additionally you may want to see about auto-installing these packages in the script or at least letting the user know these packages are needed before continuing with the installation (either in the script or readme)

Hope this helps!

antheas, 2024-08-05:
Thanks for finding the packages!

I am not interested in messing with the user's system with that script, so I'd rather them install them themselves. I added `set -e` and your packages to the readme.

Most users do not install Handheld Daemon themselves anymore as it is vendored by gaming distros. So I might have neglected that documentation a bit.

https://github.com/hhd-dev/hhd/commit/5bf8e6ca2cce929403323124ad444c271f22cdb4
https://github.com/hhd-dev/hhd/commit/c57a57ea434d3fb248f1ecd3462d19ec8daeb914

closed 2024-08-05 by antheas

kspes, 2024-08-05:
Thanks for the quick reply (and fix), glad I could help!

Great work on HHD, works really nice and was pretty easy to install/set-up :)