hhd-dev/hhd #86: HHD Update

Bladen123, 2024-08-07:
Since Last Update Controller seems to Not Work in Games anymore , feels Like a never ending cycle of Bugs and Problems with Ally and SteamOS . Works , than works Not , than works and again Not , every Update Breaks Something or makes Something Work again . I think i Switch Back to Windows. CachyOS fixed i think they Problem with suspend works now . But now i feelt Like this time this Issue comes from HHD Side with the Controller Issue.

antheas, 2024-08-07:
Hi bladen, I am sorry you are having issues with CachyOS

I have not changed how the controller works in months, so I would refer you to CachyOS for support. From looking at their kernel patches, I doubt suspend works though.

I would also like to note that the RGB issue has been fixed on Bazzite and was unrelated to Handheld Daemon. We are currently looking into fixing the bug for existing devices while waiting for a reply from Asus.

Evilexium, 2024-08-07:
Hey its me Bladen , its my Smartphone account the Bladen123. I figured it out the problem with Assassins Creed Syndicate worked with Proton GE 8-32 before after the HHD update dont know how thats possible , but i had to deactivate the Steam Input and now works . So its maybe my fault becasue i reported this as error or Bug its clear now , sometimes acts weird the steamOS if some updates are made dont know how its possible it changes some settings. For example KODI was set with a Custom Layout for controller , after the HHD update and cachyOS update the controller Layout was gone LOL. We can close this One. I apologize becasue i just see now the changes made by himself the SteamOS

closed 2024-08-07 by antheas