hhd-dev/hhd #87: Cant install on POP os - metadata issue

FinalTwist, 2024-08-20:
i run the install command and get the following error

"
  WARNING: Generating metadata for package hhd produced metadata for project name unknown. Fix your #egg=hhd fragments.
Discarding https://files.pythonhosted.org/packages/63/eb/feb42e0676260117a7d04fd40bee584e063f3ef5fa9e5d077fab0cd99dda/hhd-3.3.8.tar.gz#sha256=74e56ca35737a5a238a6cd260f606674885ab5c114e47b9928bb6b8db851d4e4 (from https://pypi.org/simple/hhd/) (requires-python:>=3.10): Requested unknown from https://files.pythonhosted.org/packages/63/eb/feb42e0676260117a7d04fd40bee584e063f3ef5fa9e5d077fab0cd99dda/hhd-3.3.8.tar.gz#sha256=74e56ca35737a5a238a6cd260f606674885ab5c114e47b9928bb6b8db851d4e4 has inconsistent name: filename has 'hhd', but metadata has 'unknown'
"
this is a brand new install of popos so i am likely missing some type of python library.  are there basic apps/repos i can try and install that may explain the above?

antheas, 2024-08-20:
Uh, use popos 24 and the Ubuntu steps should work

I think popos 22 will be too old. If you still want to use popos 22, using the deadsnakes ppa and installing python 3.12 should work

antheas, 2024-08-20:
You will need to run the install script manually and replace the python lines with python3.12

FinalTwist, 2024-08-20:
> Uh, use popos 24 and the Ubuntu steps should work
> 
> I think popos 22 will be too old. If you still want to use popos 22, using the deadsnakes ppa and installing python 3.12 should work

I wasnt even aware there was a popos 24 - from what I saw 22 was the latest.
I'll try installing python 3.12 and seeing if it will work

FinalTwist, 2024-08-20:
> I wasnt even aware there was a popos 24 - from what I saw 22 was the latest. I'll try installing python 3.12 and seeing if it will work

installed python 3.12 and stll getting this error : 
Collecting hhd
  Using cached hhd-3.3.8.tar.gz (232 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package hhd produced metadata for project name unknown. Fix your #egg=hhd fragments.
Discarding https://files.pythonhosted.org/packages/63/eb/feb42e0676260117a7d04fd40bee584e063f3ef5fa9e5d077fab0cd99dda/hhd-3.3.8.tar.gz#sha256=74e56ca35737a5a238a6cd260f606674885ab5c114e47b9928bb6b8db851d4e4 (from https://pypi.org/simple/hhd/) (requires-python:>=3.10): Requested unknown from https://files.pythonhosted.org/packages/63/eb/feb42e0676260117a7d04fd40bee584e063f3ef5fa9e5d077fab0cd99dda/hhd-3.3.8.tar.gz#sha256=74e56ca35737a5a238a6cd260f606674885ab5c114e47b9928bb6b8db851d4e4 has inconsistent name: filename has 'hhd', but metadata has 'unknown'

FinalTwist, 2024-08-20:
> You will need to run the install script manually and replace the python lines with python3.12

I am sorry I don't understand how to do this - is there a tutorial/example i could follow?

antheas, 2024-08-20:
`alias python=python3.12` then run the script 

Why are you using popos and with what device

FinalTwist, 2024-08-20:
no dice on that alias setting, I don't think python  3.12 is the issue here.  Perhaps something else which comes in ubuntu23+ which isnt in popos 22?

//
tu@tl:~$ alias python=python3.12
tu@tl:~$ curl -L https://github.com/hhd-dev/hhd/raw/master/install.sh | bash
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  2002  100  2002    0     0   3094      0 --:--:-- --:--:-- --:--:--  3094
Collecting hhd
  Using cached hhd-3.3.8.tar.gz (232 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package hhd produced metadata for project name unknown. Fix your #egg=hhd fragments.
Discarding https://files.pythonhosted.org/packages/63/eb/feb42e0676260117a7d04fd40bee584e063f3ef5fa9e5d077fab0cd99dda/hhd-3.3.8.tar.gz#sha256=74e56ca35737a5a238a6cd260f606674885ab5c114e47b9928bb6b8db851d4e4 (from https://pypi.org/simple/hhd/) (requires-python:>=3.10): Requested unknown from https://files.pythonhosted.org/packages/63/eb/feb42e0676260117a7d04fd40bee584e063f3ef5fa9e5d077fab0cd99dda/hhd-3.3.8.tar.gz#sha256=74e56ca35737a5a238a6cd260f606674885ab5c114e47b9928bb6b8db851d4e4 has inconsistent name: filename has 'hhd', but metadata has 'unknown'

antheas, 2024-08-20:
At most you will get some gyro out of it if you install it, maybe an extra button or two for emulators.

Perhaps I was wrong with alias and it does not work.

Download the install script, and replace `python3` in `python3 -m venv --system-site-packages venv` with `python3.12`. Then run it and it should work.

FinalTwist, 2024-08-20:
first off, i appreciate your help here - 

i downloaded entire code, modified install.sh for the following line
python3.12 -m venv --system-site-packages venv

and get the following error
Error: Command '['/home/tu/.local/share/hhd/venv/bin/python3.12', '-m', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1.

I have python3.12, typing that in command line lets me enter the python environment. unsure what error above is trying to tell me

antheas, 2024-08-21:
You need the packages `python3.12-venv`, `python3.12-pip` and `python3.12-dev`. It is indeed confusing if doing for the first time

antheas, 2024-10-26:
Oh well, POP_os support will come at a later time.

closed 2024-10-26 by antheas