hhd-dev/hhd #88: Pass-through Steam/View inputs

Namaneo, 2024-08-23:
Hey hey!

First of all, thanks for all the hard work, I'm using HHD on my Rog Ally and it's fantastic!

I'm trying to figure out how to pass-through the Steam and View buttons to Steam UI. When I go to the Steam controller test page, I see those behaviors (shortcuts support has been disabled in HHD settings):
* A short press to the Steam button triggers the Xbox button after a small delay. A long press doesn't trigger anything.
* A short press on the View button (the ROG one on my device) triggers the Xbox button and the A button (any idea why both?), after a little while as well. A long press switches to mouse mode (haven't found a way to disable it either, might not even be related to HHD?).

I'm assuming those are special buttons, and a special input handling is performed under the hood to translate them to something Steam would understand. Is there a way to disable any extra processing, and just forward down and up events like for every other button? That'd make the UI interactions much snappier, and HHD would still be accessible through the extra keys or the `hhd-decky` plugin.

Thanks!

antheas, 2024-08-23:
What's your device?

Half of the extra delay for QAM happens to support double tap and half of it because Xbox + A requires a delay

We are trying to remove Xbox + a and move to Ctrl+2 but that requires patching gamescope and the patch for that still needs tuning https://github.com/ValveSoftware/gamescope/pull/1430

As for the double tap delay, yes if you disable the shortcuts functionality it disappears already

Xbox button should not have a delay. If it does you probably have an ally where in that case, Asus would have to give us something better

antheas, 2024-08-23:
There was an option to disable the QAM handling entirely and emit the button as the dualsense mute button but people used to mess with it and break the overlay so I removed it

Namaneo, 2024-08-23:
Thanks for answering, I indeed have a ROG Ally (Z1E) running Bazzite. So I understand the delays I'm seeing is solely because of Asus being weird with the associated input events?

Also, and that's just out of curiosity, is there a reason why the emulated controller device can't be an actual Steamdeck, with its specific input events/codes for the Steam and View buttons?

(Closing for now as there's nothing to do at the moment, but will keep an eye on that gamescope PR!)

closed 2024-08-23 by Namaneo

antheas, 2024-08-23:
The delay on the Xbox button yes. The delay on qam is that + the double tap + the XboxA delay, which gets a bit much

antheas, 2024-08-23:
As for emulating a steam deck, valve uses a different proprietary backend for that that opens it as a USB device.

So we would have to emulate a USB device which would be a pain + it has a lot of buttons not available in other devices so mappings would be confusing for most users

And it would only solve the final 150ms

antheas, 2024-08-23:
The decky plugin is deprecated and has started breaking now, so the double tap is kind of the only way now. But perhaps with touch gestures it can be made optional now