hhd-dev/hhd #89: Sleep detection seems to be broken

nemanjan00, 2024-08-23:
Hello,

My journal seems to be full of these messages:

```
Aug 23 15:53:31 pocket hhd[1020]: MAIN   INFO     System woke up from sleep. Wakeup count: 2128 from 2112.
Aug 23 15:53:31 pocket hhd[1020]: SMUQ   INFO     Waking up from sleep, resetting TDP after 4 seconds.
Aug 23 15:53:33 pocket hhd[1020]: MAIN   INFO     System woke up from sleep. Wakeup count: 2137 from 2128.
Aug 23 15:53:33 pocket hhd[1020]: SMUQ   INFO     Waking up from sleep, resetting TDP after 4 seconds.
```

I can confirm system is not going to sleep....

antheas, 2024-08-23:
Can you find what's increasing your wake count?

nemanjan00, 2024-08-23:
Apparently, it is serio0...

I guess I am going to have to track down why my keyboard driver keeps incrementing wakeup count...

antheas, 2024-08-23:
Would be good to find out if there are certain devices causing this and or find an alternative

nemanjan00, 2024-08-23:
Ok, apparently, I am not alone...

https://www.reddit.com/r/gpdwin/comments/1anhxjf/info_random_wakeups_on_sleep_on_linux_have_been/

I do have this in log, for that device

```
[    9.332782] acpi PNP0C50:00: Ignoring wakeup on pin 8
```

However looks like it is still happening. 

I know this is out of scope for you, but, I am writing just in case someone else has the same issue

antheas, 2024-08-23:
Lets just say that reading the wakeup count is not exactly conventional

nemanjan00, 2024-08-23:
Ok, update

```
root@pocket: /sys # cat ./devices/platform/i8042/serio0/power/wakeup_active
0
root@pocket: /sys # cat ./devices/platform/i8042/serio0/power/wakeup_active_count 
5503
```

This devices is sending wakeup events, but they should be ignored, as I understand the docs

https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-devices-power

nemanjan00, 2024-08-23:
This could be interesting source of information:
```
cat /sys/kernel/debug/wakeup_sources
```

nemanjan00, 2024-08-23:
Any workaround you can recommend?

antheas, 2024-08-23:
Yeah, dont look at journalctl for a few days as I find a workaround :)

antheas, 2024-08-23:
If you are on arch, you can just downgrade to 3.3.7. Just mind that TDP might not be set correctly after sleep.

nemanjan00, 2024-08-23:
I would ignore it, but, I was afraid it might be impacting performance by constantly sending ACPI to change TDP

Downgraded to 3.3.7 and locked update, until you close this issue

antheas, 2024-08-23:
If it set the TDP, you would have gotten logs about that. It queues a change after 4 seconds, in perpetuity...

antheas, 2024-08-23:
Try using the git variant https://github.com/hhd-dev/hhd/commit/c57c45c8014ac265eed213ac3ee03046c2b16c19

nemanjan00, 2024-08-23:
Looking at the log, it does seem it has set it quite a few times:

```
Aug 23 19:08:35 pocket hhd[1054]: SMUQ   INFO     Waking up from sleep, resetting TDP after 4 seconds.
Aug 23 19:08:40 pocket hhd[1054]: ASMU   INFO     Sending SMU command with 4 parameters:
Aug 23 19:08:40 pocket hhd[1054]:                  -   fast_limit (0x06): 41
Aug 23 19:08:40 pocket hhd[1054]:                  -   slow_limit (0x07): 32
Aug 23 19:08:40 pocket hhd[1054]:                  -   skin_limit (0x2e): 30
Aug 23 19:08:40 pocket hhd[1054]:                  -  stapm_limit (0x05): 30
Aug 23 19:08:40 pocket hhd[1054]: ASMU   INFO     Executing ACPI call:
Aug 23 19:08:40 pocket hhd[1054]:                 '\_SB.ALIB 0x0c b16000628a0000007007d00002e307500000530750000'
```

nemanjan00, 2024-08-23:
I use system package manager to install hhd, so it was nontrivial for me to update to specific commit.

I applied changes from that commit, so, it looks it might be ok, but, I just noticed another error and I am unsure if it is related

```
Aug 23 20:18:22 pocket hhd[5460]: Traceback (most recent call last):
Aug 23 20:18:22 pocket hhd[5460]:   File "<frozen runpy>", line 198, in _run_module_as_main
Aug 23 20:18:22 pocket hhd[5460]:   File "<frozen runpy>", line 88, in _run_code
Aug 23 20:18:22 pocket hhd[5460]:   File "/usr/lib/python3.12/site-packages/adjustor/drivers/amd/ppd.py", line 169, in <module>
Aug 23 20:18:22 pocket hhd[5460]:     name = dbus.service.BusName(iface(legacy), session_bus)
Aug 23 20:18:22 pocket hhd[5460]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 23 20:18:22 pocket hhd[5460]:   File "/usr/lib/python3.12/site-packages/dbus/service.py", line 134, in __new__
Aug 23 20:18:22 pocket hhd[5460]:     retval = bus.request_name(name, name_flags)
Aug 23 20:18:22 pocket hhd[5460]:              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 23 20:18:22 pocket hhd[5460]:   File "/usr/lib/python3.12/site-packages/dbus/bus.py", line 299, in request_name
Aug 23 20:18:22 pocket hhd[5460]:     return self.call_blocking(BUS_DAEMON_NAME, BUS_DAEMON_PATH,
Aug 23 20:18:22 pocket hhd[5460]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 23 20:18:22 pocket hhd[5460]:   File "/usr/lib/python3.12/site-packages/dbus/connection.py", line 634, in call_blocking
Aug 23 20:18:22 pocket hhd[5460]:     reply_message = self.send_message_with_reply_and_block(
Aug 23 20:18:22 pocket hhd[5460]:                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Aug 23 20:18:22 pocket hhd[5460]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.AccessDenied: Request to own name refused by policy
```

antheas, 2024-08-23:
That is "ok". It is caused by you not having installed the power-profiles-daemon package, so PPD emulation is not working.

The packaging around that policy file is not worked out yet.

antheas, 2024-08-23:
Try `hhd-git` or use the local installer, which makes it possible to use the git version. You can have both the local and package variants installed at the same time. Just take some care in only having one service active at a time.

nemanjan00, 2024-08-23:
Ok, replaced tlp with ppd...

Installed hhd-git and looks like I do not have that issue anymore

antheas, 2024-08-23:
You will probably have to mask PPD too on arch. In any case, handheld daemon controls the important CPU settings based on TDP. It does not replace the nice wifi tlp when docking feature though.

So, test a bit more and then close?

Also test if the tdp is set correctly upon resume.

nemanjan00, 2024-08-23:
Are tlp and hhd in conflict?

Tests look good. It did change TDP after resume

nemanjan00, 2024-08-23:
Thanks for info

closed 2024-08-23 by nemanjan00