hhd-dev/hhd #90: Antec Core HS missing 2 Xbox buttons

clairearmstrong, 2024-08-27:
I am using the KDE Bazzite environment on the Antec Core HS, which is a rebranded Ayaneo Slide.
The Xbox Home and Share buttons are not mapped to anything. There are 3 spare buttons on the Antec that don't seem to do anything on Bazzite

antheas, 2024-08-27:
`cat /sys/devices/virtual/dmi/id/product_name`

clairearmstrong, 2024-08-27:
SLIDE

antheas, 2024-08-27:
`journalctl -b | fpaste`

They really did not even change the product name

antheas, 2024-08-27:
Ok, here is the command to fix:

```
sudo systemctl enable --now hhd@$(whoami)
```

The slide dmi was missing from Bazzite. It will be added. You can run this command and after verifying everything works close the issue.

clairearmstrong, 2024-08-27:
I have the RGB on the thumbsticks turned off, but now when I sleep it in Steam the RGB comes on, and the fan is still running. I can't say for definite but I'm fairly sure the fan was off when I slept it earlier today

antheas, 2024-08-27:
Welcome to the Linux experience on Ayaneo devices. 

You can run the previous command with disable then see if it still happens. Although I am going to be honest with you, the Slide has not been a good experience on Linux.

closed 2024-08-28 by clairearmstrong

antheas, 2024-08-28:
I will probably add a note in the readme so that people are not misled by the supported list meaning linux support.