hhd-dev/hhd #91: Steam key shortcuts doesn't work right on Dualsense emulation

Nman9000, 2024-08-28:
On the legion go running bazzite, when emulating either dualsense or dualsense edge using the left legion button and the right trigger to take screenshots it will zoom in then take the screenshot. The left trigger also does the same thing as the right trigger and the left thumbstick doesn't control the volume. When switched to emulate the xbox controller it works normal and as expected.

antheas, 2024-08-28:
Can you verify this in SteamUI? On a per game basis, all mappings might be affected by the current gamepad profile.

Nman9000, 2024-08-28:
On the steam home screen, the left thumbstick works for volume but both triggers still take screenshots, neither will magnify but that seems to just be the way it works there. I have also tested this in a few games and it all does the same thing when emulating either dualsense. Double checked in steam controller test device inputs and it shows that each trigger only activates their respective input and not each others.

antheas, 2024-08-28:
Seems like a steam input mapping is the cause. I think the issue might be steam or a community mapping

Are you on the stable or beta?

Nman9000, 2024-08-28:
Im on stable

Nman9000, 2024-08-28:
Another thing i found is if i hold the legion key until the list of shortcuts comes up then hit the right trigger it will just take a screenshot or if i hit the left trigger it will just magnify. Guess it could be steam, i tried using an empty template in a game and it still does it so i don't think it is a community layout.

antheas, 2024-10-26:
I think that Dualsense might have different mappings now. But in any case, with the new Steam Controller emulation, you should get gyro + correct glyphs + correct shortcuts.

closed 2024-10-26 by antheas