hhd-dev/hhd #92: GPD Win Mini Fan Control Support

SimpleHeuristics, 2024-08-29:
the GPD win mini supports fan control through the power control decky plugin, however the HHD interface for TDP control is much better but cannot co-exist with the power control plugin. Is I possible to add support for win mini fan control?

antheas, 2024-08-30:
Fan control for ayaneo and gpd devices is on the TODO.

This is because unlike the ROG Ally and Legion Go they do not have an embedded fan curve adjustment.

Instead, the fan speed needs to be set periodically by a program, in this case hhd.

In order to do that I need access to a device that has a software fan curve that can be tuned. After that it should be possible to rollout on all devices.

SimpleHeuristics, 2024-08-30:
Is it possible for me to assist with my device?

antheas, 2024-08-30:
Unfortunately, not at this time. I will leave the issue open for when i get to it.