hhd-dev/hhd #94: legion go fan curve not working

samuelwang456, 2024-09-04:
I have tried bazzite, cachyos, nobora, currently I keep in bazzite, all encountered the same problem, custom fan curve does not work.
When the custom fan curve is disabled, the fan runs normally, 
once the custom fan curve is turned on and the speed is set to the speed I want, the fan suddenly drops to a very low speed, lower than the speed I set, and even when the CPU temperature rises, the speed does not increase, It keeps the RPM very low, it doesn't grow.
I have upgraded to the latest version of bios，Can anyone give me some help, I do enjoy using linux as a gaming system, but the default RPM of legion go is too noisy for me to use properly

antheas, 2024-09-04:
This bug happens when using too low of a fan speed

It's related to the firmware of the device

Try to be more conservative or use the device presets that have better fan curves

samuelwang456, 2024-09-04:
> This bug happens when using too low of a fan speed
> 
> It's related to the firmware of the device
> 
> Try to be more conservative or use the device presets that have better fan curves
> 
> The legion go does not have a powerful fan so you will need to get used to it

Thank you for your reply. I finally understand the reason. Thank you.
Maybe I should just turn the revs down a little bit, not down too much

antheas, 2024-09-04:
Yep

closed 2024-09-04 by antheas