hhd-dev/hhd #95: HHD is only controlling the CPU's tdp, not the GPU

bbgoose92, 2024-09-07:
I'm on Arch Linux with the ACPI kernel module installed on the ROG Ally X. 
setting the tdp to 10w for example will cause the CPU to stay under 10w, but the GPU will do what ever it pleases. 
is there some other requirement to it controlling the GPU's tdp?

antheas, 2024-09-07:
The acpi module is not used on the ally, at least for now

The requirement is Asus wmi. However, Asus wmi cannot load due to a conflict with the amd pmf module, specifically on the Ally X.

This is fixed with an upcoming kernel patch that will probably land on 6.12. However, you may opt to block it for now.

The minimum kernel version for the ally x is 6.11 if you want the controllers to come out of suspend. Otherwise, you need to backport the quirk patch from 6.11.

bbgoose92, 2024-09-07:
Okay awesome thank you for the quick response, and for your time.

closed 2024-09-07 by bbgoose92

Sid127, 2024-11-04:
do we know if this patch has landed in 6.12?