hhd-dev/hhd #96: Double Input with Keyboard and gamepad Rog Ally OG paddles

WillySou, 2024-09-08:
Hi,

For starters thanks a lot for your great work 👏 it's really incredible what you achieved here. 

Now for my issue:
Lately I've been having a problem that happens when I press the back paddles in-game. It sends a mix of keyboard and gamepad signals that confuses many games like spiderman, God of war, Hogwarts Legacy. I use the paddle as a turbo simple attack button to avoid the usual button mashing.
I will send you a video showing this. I'm on latest build I think.
Thanks a lot! 

https://github.com/user-attachments/assets/a3fce5f9-f302-46d5-bc06-a7aa3eeb2091

antheas, 2024-09-08:
Nice find, I will think about how to fix it soon.

This is caused by not muting the keyboard device that is used for shortcuts, since it is re-used by mouse mode.

Weird how nobody caught it yet. My Ally is stuck in customs.

WillySou, 2024-09-08:
@antheas thanks a lot 👍 I won't use my Ally for the next week because I will travel a lot and will bring my Legion Go instead. But when I come back I will be glad to help 👍

antheas, 2024-09-12:
https://github.com/hhd-dev/hhd/releases/tag/v3.3.14 fixed

closed 2024-09-12 by antheas