hhd-dev/hhd #99: [Feature Request] TDP profile based on power source

dstrnad, 2024-09-29:
Is it possible to add an option in Handheld Daemon (HHD overlay preferably) to automatically switch power profiles based on the power source (AC or battery)? For example, when disconnecting from the charger, it would automatically switch to Silent mode, and when connecting to the charger, it would switch to Performance mode, similar to how this works in Armoury Crate on Windows. 
It would be also great to have two custom TDP presets. For example, on the ROG Ally, it's very convenient to automatically switch to a 17/18W TDP on battery and a full 25/30W TDP when plugged into AC power.

antheas, 2024-09-30:
Doing some form of presets is high on the priority list. It will happen soon, just do not know how soon that soon is.

SeeDborg, 2024-12-20:
Any update on this ? It would be very interesting 😄