hhd-dev/hhd #101: Rog Ally, CPU stuck at 544mhz after wake from sleep

dstrnad, 2024-10-08:
I’m using a original ROG Ally with the Z1 Extreme chip, running the latest version of Bazzite. Occasionally, after waking the device from sleep mode, the CPU gets stuck at a low frequency—around 544MHz as shown in the Steam overlay. Adjusting the TDP through Handheld Daemon or Steam’s overlay doesn’t resolve the issue; the system remains slow until I perform a restart. The issue appears randomly and doesn’t happen every time.

I suspect it might be related to the charge limit I’ve set in Handheld Daemon, as I don’t recall encountering this problem when the limit was set to 100%. Currently, it’s set to 80%, but I’m not sure if this is directly causing the issue, it may be a coincidence..

Here is a reference to people having the same issue:
https://www.reddit.com/r/LinuxOnAlly/s/afj6v3slOj
https://www.reddit.com/r/LinuxOnAlly/s/JzcRrJkKRK

I also created the same bug in bazzite github repo, as Im not sure if HHD can fix this.

antheas, 2024-10-08:
Can you rebase into bazzite testing and see if you can reproduce it?

antheas, 2024-10-08:
We are testing a patch that might fix it right now.

dstrnad, 2024-10-14:
> Can you rebase into bazzite testing and see if you can reproduce it?

Do you mean that the fix is availabe in a newer HHD version which will be downloaded when I change bazzite’s update branch to (:testing)?  Which exact version of HHD contains the fix? Thanks

dstrnad, 2024-10-24:
Testing with the current stable version and HHD 3.5.6 shows the issue occurs much less frequently, but it did happen again today. With a custom TDP set to 25W and no TDP boost, after waking up the Ally, it got stuck at 544MHz/1099MHz once more.

antheas, 2024-10-24:
So you had it happen today. Has it happened again? Did you need to reboot or did it fix itself?

Do you still have windows? If yes, can you go to windows and disable modern standby assistant in armoury crate? And update your bios for good measure

dstrnad, 2024-10-24:
It has already happened twice today, and yes, a reboot is needed to fix it—changing power profiles doesn’t help. Modern standby is disabled in Windows, and the BIOS is up to date (v341). Both times it occurred in the same scenario: I was gaming in docked mode using the CEMU emulator, put the system to sleep, waited a few hours, and then woke it up while still docked, without disconnecting in between. On both occasions, the CPU was stuck at 544/1099 MHz, according to the overlay.

antheas, 2024-10-24:
Yes, the only scenario remaining where it happens is when people are docked for multiple hours.

antheas, 2024-10-24:
https://gitlab.freedesktop.org/drm/amd/-/blob/master/scripts/amd_s2idle.py you need to run this script with a long enough delay and make it happen, then drop the log here

I need to see the wakeup that happens when docked and breaks the CPU so I can look more into it