hhd-dev/hhd #102: Can't Map Back Buttons Ally X Bazzite

tehss1, 2024-10-11:
I was using the xbox elite option for HHD within Bazzite to map the back buttons on my ally x. It looks like xbox elite and dualsense edge are no longer mappable in HHD after the latest update. I just have options for Xbox or Dualsense. Is there another way to map the back buttons I'm missing or is this a bug in that I can no longer manually select Xbox Elite? Thanks!

antheas, 2024-10-11:
That should not be the case... there has not been an update in Bazzite in 11 days.

If you are indeed seeing this you should also be seeing the "Steam Controller" option and be in testing.

If that is the case, dualsense now acts as dualsense edge by default and Xbox Elite has been replaced by Steam Controller

tehss1, 2024-10-11:
Thank you! I'm on the testing branch and can confirm Steam Controller is listed. I've selected steam controller and enabled the back buttons in the controller menu per game but they aren't working.

antheas, 2024-10-11:
Check if you can see them in the controller tester, if not that is a bazzite issue we will have to look into.

tehss1, 2024-10-11:
controller tester shows "false" for L4/R4. let me know if there's anything I can do to help test/provide information.

antheas, 2024-10-11:
This is weird. Mine work. Wait until the next testing build and we will see. Do not update to the current one, 6.11 broke the speakers of the ally and there are other issues. Perhaps one of those is that.

tehss1, 2024-10-11:
I noticed my speakers stopped working also 😂 I’ll wait for the next
release and see what happens. Thanks again.

tehss1, 2024-10-15:
all set on the latest release, still having some trouble with the steam controller but all is well with xbox and the elite option added back. thanks!

closed 2024-10-15 by tehss1