hhd-dev/hhd #109: Phantom X button press on ROG Ally X

memmam, 2024-10-26:
Hello!

I have a ROG Ally X, on which I've installed Nobara and the ROG Ally / Ally X-specific kernel found at https://github.com/jlobue10/ALLY_Nobara_stuff. I've uninstalled the included inputplumber installation and replaced it with HHD.

My problem is, when I press the Armoury Crate button, it seems to produce phantom button presses.

One press, and it presses X (A) on whatever screen I'm on before it brings up the QAM.

Two presses, and it does the above behavior, and then also seems to press X (A) _again_ in the HHD menu, usually opening the TDP menu.

I'm not sure if this is a problem with HHD or with my specific setup, but I can't find a community Discord or any other support mechanism for HHD, and I can't find any relevant settings looking at the config file, so I'm opting to post here hoping that my issue can be fixed, and if it's a deeper issue with HHD, it can be fixed for others.

Thank you! If there's any other information I can or should provide, please let me know; I'll be watching this issue for any replies.

antheas, 2024-10-26:
If you want to use nobara, use [this kernel](https://github.com/hhd-dev/linux-bazzite), remove hhd from dnf, and install either the copr or local install variants from here.

Also, the kernel above uses a new kernel patch that automatically selects the correct Ally speaker firmware, so you will want to make sure your linux-firmware is up to date, in case GE messes with that too.

Otherwise, nobara is unfortunately unsupported.

antheas, 2024-10-26:
You are right though, I need to stick a discord link somewhere.

memmam, 2024-10-26:
Switching kernels did in fact fix the button press issue, but now I don't have any audio. Nobara reports linux-firmware as version 20240909-1.fc40.noarch.

I found the list of firmware files contained within linux-firmware on git.kernel.org. Can you specify which firmware file I need?

antheas, 2024-10-26:
You need the Linux firmware that came last week, 17.

Otherwise https://gitlab.com/kernel-firmware/linux-firmware/-/commit/4cbaca8d279c5b37dc999e9e6b9a5d8c3cbad2f5

antheas, 2024-10-26:
Place on /usr/lib/firmware not in a subdir

memmam, 2024-10-26:
Can confirm all issues are solved after copying those two files! It IS a shame that I had to switch away from the kernel I was using, as it contained Zen 4-specific link-time optimizations, which I assume this Bazzite kernel lacks, but c'est la vie.

Thank you for your assistance. :)

closed 2024-10-26 by memmam

antheas, 2024-10-26:
The bazzite kernel ships with sched ext and BORE, so even though it does not use LTO it will be better.

As for LTO, there are not many benchmarks showing that it's better and it makes compilation take 3x longer.

memmam, 2024-10-26:
Fair enough! Thank you again for your assistance, and sorry for hassling you about this.

memmam, 2024-10-26:
@antheas for future reference, I ended up having new, worse, unexplained problems with HHD gradually cropping up out of nowhere, I'm just biting the bullet and switching to Bazzite even though I was trying to avoid it. I think it's safe to say Nobara is like, mega super turbo unsupported.

antheas, 2024-10-26:
Would be good to list them.

memmam, 2024-10-26:
> Would be good to list them.

Okay so issues I ran into not documented already in this ticket are:

1. Nobara insists on reinstalling the included Decky plugins on reboot. This includes SimpleDeckyTDP, which is incompatible with HHD. This can be resolved by removing SimpleDeckyTDP from `/usr/share/deckyloader` in addition to from `~/homebrew`.
2. I stopped being able to use the B button to back out of the Steam keyboard from within Desktop Mode, for some reason?
3. After a while my button bindings got really messed up (Triangle became Square, L1 became Triangle, Start seems unbound entirely, probably other things I haven't noticed)
4. This may have been intended behavior / user error, but when I reinstalled HHD and switched kernels, shortcut support unchecked itself which was frustrating until I figured it out

antheas, 2024-10-26:
4) I think nobara force reinstalls inputplumber every time you update, and maybe removed hhd

For 1-3, ack. Seem unrelated though.

Ok so its safe to call nobara unsupported moving forward.

memmam, 2024-10-26:
2 and 3 only started after switching kernels, so I think it's safe to say that since HHD has issues both with the Nobara kernel and the Bazzite kernel, yeah, probably wise.

Ideally if you could solve the phantom button input issue with the Nobara kernel it'd be fine, but that seems spectacularly out-of-scope.

antheas, 2024-10-26:
Got my numbers confused, its probably inputplumber, it would not just do SimpleDeckyTDP.

memmam, 2024-10-26:
Just checked with dnf and it reports inputplumber is still uninstalled. I'm not blaming the Bazzite kernel but something about switching to the Bazzite kernel has made Nobara+HHD _very_ angry, if I thought it was a Bazzite kernel issue I wouldn't be planning to switch to Bazzite + the kernel you linked.

antheas, 2024-10-26:
My Ally X is currently on Bazzite and I do most of my dev work on Bazzite now.

I cant tell you what caused what you are experiencing (without logs; `journalctl -b | fpaste`) and perhaps its not worth finding out either.

memmam, 2024-10-26:
Agreed, just wanted to make you aware of it before I nuke my Nobara install into the dirt and install Bazzite so you can tell people in the future not to bother. I don't think diagnosing this further is really worth either of our time.