hhd-dev/hhd #111: Handle situations when CONFIG_DNOTIFY is unavailable

waffleabyssx0x, 2024-10-28:
Hello, I am using a kernel which is not having CONFIG_DNOTIFY set. When running HHD, the following lines cause HDD to not load, and exit with an error. Commenting out these lets HHD start, and run successfully, however I am not sure about the consequences of running it without this piece of code. I would like to work around this issue, maybe have a warning that CONFIG_DNOTIFY is unset and let the code run anyway without this feature - or use a different method to achieve the same functionality.
Thank you.

https://github.com/hhd-dev/hhd/blob/9d7bf94c9cc5c07f076305f23a7dbe6ef4dd68dc/src/hhd/__main__.py#L414-L425

antheas, 2024-10-28:
It is used to allow configuration hot reloading. Perhaps you should enable it in your kernel.

I can try catch it too as it's optional

antheas, 2024-11-28:
https://github.com/hhd-dev/hhd/commit/aa9745298d0a3995f76ba5bcf7703cde0fd9cbd0

closed 2024-11-28 by antheas