hhd-dev/hhd #113: GPD Win Mini 2024 - missing Manual GPU Control in QAM

georgepauna, 2024-11-07:
Manual GPU Control in QAM is missing and it seems the entry is there but hidden, because when i am on TDP Limit and I press down the performance area scrolls to the top, and when I press down again, it goes back and continues to Scaling mode.

antheas, 2024-11-07:
Unfortunately, the current Steam slider is limited to Steam Deck clock values which makes it useless for other devices. It also only appears if there is a positive result on the polkit.

So the decision was made to hide it by denying the polkit access to the frequency values in Bazzite. If you re-enable it it will work. There are decky plugins that used to change the values.

closed 2024-11-07 by antheas

antheas, 2024-11-07:
I did not know that QAM navigation is broken. It shouldnt be unless you enable the slider. I will check it later today. There used to be an env variable for it too.

georgepauna, 2024-11-07:
> I did not know that QAM navigation is broken. It shouldnt be unless you enable the slider. I will check it later today. There used to be an env variable for it too.

I don't even see the entry with the checkbox, to enable the slider. Thanks a lot for looking into this!

antheas, 2024-11-07:
It works fine on Bazzite with navigation too, regardless of if the slider is enabled. 

In any case, this is something that is particular to your setup if you cannot see the checkbox. I am pretty sure you always can. Unless something changed in the Steam Beta.