hhd-dev/hhd #115: HHD 3.6.0 Gamescope Poweroff Screen before Sleep GPD Win Mini 2024 keeps screen off even when waking from sleep.

SimpleHeuristics, 2024-11-19:
New updated of HHD with Bazzite Testing, the poweroff screen option turns of the screen with sleeping, but when waking the device the screen does not turn on again. Audio can be heard and it appears to UI can be navigated.

Ctrl + F2 works to get into a TTY and screen turns on again but then can't return to the game scope session.

antheas, 2024-11-19:
`journalctl -b | fpaste` after it happens

SimpleHeuristics, 2024-11-19:
https://paste.centos.org/view/323e56b7

antheas, 2024-11-19:
Looks like it is working. If you disable the setting does it work? 

you also have an oops during sleep that might have broken the display

SimpleHeuristics, 2024-11-19:
Yeah it works fine with the setting disable. sleep / suspend always worked on the win mini without issue before. The screen turned off when sleeping and right back on when waking.  It's just when this setting got ticked on that after sleep the display doesn't turn on. The display work for a TTY session but can't switch back to the steam session.with the setting disabled it works the same as before.

antheas, 2024-11-19:
I tried my ally, win 4, and Legion go and they all worked fine :(

antheas, 2024-11-20:
Ok, try turning off the feature and then doing the following (ideally over ssh) manually to test it out

```
hhd.contrib gamescope :0 GAMESCOPE_DPMS=0
hhd.contrib gamescope :0 GAMESCOPE_DPMS=1
```

You might need to use `:1` instead

Then sleep the device inbetween as well.

SimpleHeuristics, 2024-11-21:
So press the power button to sleep between the two commands?

antheas, 2024-11-21:
Sure, and see if the commands work as expected.

SimpleHeuristics, 2024-11-21:
So the first command with DPMS=0 while the screen is on didn't do anything. And then the second command DMPS=1 turned off the display while everything was still running (could hear UI noises while moving the joystick). Did DPMS=0 again and nothing happening even though in the terminal it outputs that the value was set successfully. I then slept the device and resumed it and then tried the above commands and same thing, can only hear UI sounds when moving the sticks but the screen does not turn back on. starting a TTY with Ctrl alt F2 still works. but trying to go back to game scope session with Ctrl Alt F1 does not work, just ui sounds in the background with a black screen

Here is the paste: https://paste.centos.org/view/89699342

antheas, 2024-11-22:
```
Nov 21 18:55:00 GPD-WIN-MINI kernel: ------------[ cut here ]------------
Nov 21 18:55:00 GPD-WIN-MINI kernel: WARNING: CPU: 5 PID: 5124 at drivers/gpu/drm/amd/amdgpu/../display/amdgpu_dm/amdgpu_dm.c:8747 amdgpu_dm_atomic_commit_tail+0x395e/0x4530 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel: Modules linked in: rfcomm snd_seq_dummy snd_hrtimer uhid acpi_call nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject nft_ct nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nf_tables qrtr bnep sunrpc binfmt_misc iwlmvm snd_hda_codec_realtek vfat spd5118 snd_hda_codec_generic intel_rapl_msr wl(PO) fat snd_hda_codec_hdmi snd_hda_scodec_component mac80211 amd_atl intel_rapl_common edac_mce_amd snd_hda_intel libarc4 snd_intel_dspcfg kvm_amd snd_intel_sdw_acpi btusb snd_hda_codec btrtl kvm snd_hda_core btintel iwlwifi snd_hwdep btbcm snd_seq btmtk bmi260_i2c(O) rapl wmi_bmof pcspkr bmi260_core(O) bluetooth thunderbolt snd_seq_device xpad cfg80211 bmi160_spi snd_pcm i2c_piix4 k10temp i2c_smbus regmap_spi rfkill snd_timer snd amd_pmf soundcore bmi160_i2c legion_laptop bmi160_core amdtee industrialio_triggered_buffer amd_sfh kfifo_buf tee industrialio joydev platform_profile amd_pmc tcp_bbr
Nov 21 18:55:00 GPD-WIN-MINI kernel:  sch_fq vhba(O) ryzen_smu(O) hid_playstation led_class_multicolor ff_memless loop nfnetlink lz4 lz4_compress zram amdgpu amdxcp drm_exec gpu_sched drm_buddy i2c_algo_bit drm_suballoc_helper drm_display_helper cec drm_ttm_helper ttm crct10dif_pclmul crc32_pclmul crc32c_intel polyval_clmulni polyval_generic nvme nvme_core ghash_clmulni_intel video sha512_ssse3 hid_multitouch sha256_ssse3 sha1_ssse3 sp5100_tco nvme_auth wmi i2c_hid_acpi i2c_hid serio_raw uas usb_storage v4l2loopback(O) videodev mc scsi_dh_rdac scsi_dh_emc scsi_dh_alua ip6_tables ip_tables kvmfr(O) gcadapter_oc(O) fuse i2c_dev dm_multipath
Nov 21 18:55:00 GPD-WIN-MINI kernel: Unloaded tainted modules: nct6687(O):2
Nov 21 18:55:00 GPD-WIN-MINI kernel: CPU: 5 UID: 0 PID: 5124 Comm: kworker/u64:23 Tainted: P        W  O       6.11.9-301.bazzite.fc41.x86_64 #1
Nov 21 18:55:00 GPD-WIN-MINI kernel: Tainted: [P]=PROPRIETARY_MODULE, [W]=WARN, [O]=OOT_MODULE
Nov 21 18:55:00 GPD-WIN-MINI kernel: Hardware name: GPD G1617-01/G1617-01, BIOS 1.66 04/17/2024
Nov 21 18:55:00 GPD-WIN-MINI kernel: Workqueue: async async_run_entry_fn
Nov 21 18:55:00 GPD-WIN-MINI kernel: RIP: 0010:amdgpu_dm_atomic_commit_tail+0x395e/0x4530 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel: Code: c0 0f 85 05 ce ff ff e9 60 79 57 00 48 83 fe 07 0f 83 21 08 00 00 48 69 f6 68 0f 00 00 49 8d 84 33 08 03 00 00 e9 79 eb ff ff <0f> 0b e9 c7 d6 ff ff 0f 0b e9 b0 d6 ff ff 0f 0b e9 80 d1 ff ff 48
Nov 21 18:55:00 GPD-WIN-MINI kernel: RSP: 0018:ffffa1ed477bf940 EFLAGS: 00010282
Nov 21 18:55:00 GPD-WIN-MINI kernel: RAX: 00000000ffffffea RBX: ffff9533c88c6000 RCX: 0000000000000000
Nov 21 18:55:00 GPD-WIN-MINI kernel: RDX: 0000000000000002 RSI: 0000000000000297 RDI: ffff9533e180015c
Nov 21 18:55:00 GPD-WIN-MINI kernel: RBP: ffffa1ed477bfb90 R08: ffff9533e1800010 R09: 0000000000000000
Nov 21 18:55:00 GPD-WIN-MINI kernel: R10: ffff9533d569f400 R11: ffff9533c1cf0a00 R12: ffff9533c1cf1200
Nov 21 18:55:00 GPD-WIN-MINI kernel: R13: 0000000000000000 R14: 0000000000000000 R15: ffff9533c5b9da00
Nov 21 18:55:00 GPD-WIN-MINI kernel: FS:  0000000000000000(0000) GS:ffff953905c80000(0000) knlGS:0000000000000000
Nov 21 18:55:00 GPD-WIN-MINI kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Nov 21 18:55:00 GPD-WIN-MINI kernel: CR2: 000055a9ae0b9000 CR3: 00000002ce42c000 CR4: 0000000000f50ef0
Nov 21 18:55:00 GPD-WIN-MINI kernel: PKRU: 55555554
Nov 21 18:55:00 GPD-WIN-MINI kernel: Call Trace:
Nov 21 18:55:00 GPD-WIN-MINI kernel:  <TASK>
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? amdgpu_dm_atomic_commit_tail+0x395e/0x4530 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __warn.cold+0x8e/0xe8
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? amdgpu_dm_atomic_commit_tail+0x395e/0x4530 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? report_bug+0xff/0x140
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? handle_bug+0x58/0x90
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? exc_invalid_op+0x17/0x70
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? asm_exc_invalid_op+0x1a/0x20
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? amdgpu_dm_atomic_commit_tail+0x395e/0x4530 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __free_pages_ok+0x138/0x440
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? free_unref_page_commit+0x94/0x360
Nov 21 18:55:00 GPD-WIN-MINI kernel:  commit_tail+0xac/0x160
Nov 21 18:55:00 GPD-WIN-MINI kernel:  drm_atomic_helper_commit+0x11a/0x140
Nov 21 18:55:00 GPD-WIN-MINI kernel:  drm_atomic_commit+0xa6/0xe0
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __pfx___drm_printfn_info+0x10/0x10
Nov 21 18:55:00 GPD-WIN-MINI kernel:  drm_atomic_helper_commit_duplicated_state+0xde/0xf0
Nov 21 18:55:00 GPD-WIN-MINI kernel:  drm_atomic_helper_resume+0xa5/0x160
Nov 21 18:55:00 GPD-WIN-MINI kernel:  dm_resume+0x390/0x9d0 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  amdgpu_device_ip_resume_phase2+0xa0/0x1d0 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  amdgpu_device_resume+0x7c/0x2b0 [amdgpu]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __pfx_pci_pm_resume+0x10/0x10
Nov 21 18:55:00 GPD-WIN-MINI kernel:  amdgpu_pmops_resume+0x4a/0x80 [amdgpu]
[... cut ...]
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __pfx_worker_thread+0x10/0x10
Nov 21 18:55:00 GPD-WIN-MINI kernel:  kthread+0xcf/0x100
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __pfx_kthread+0x10/0x10
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ret_from_fork+0x31/0x50
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ? __pfx_kthread+0x10/0x10
Nov 21 18:55:00 GPD-WIN-MINI kernel:  ret_from_fork_asm+0x1a/0x30
Nov 21 18:55:00 GPD-WIN-MINI kernel:  </TASK>
Nov 21 18:55:00 GPD-WIN-MINI kernel: ---[ end trace 0000000000000000 ]---
```
Let me put this here

So it never woke up again.

SimpleHeuristics, 2024-11-22:
I'm not sure how to interpret these logs but if it's referring to the display or interference with the display some how then yeah. Everything else wakes up and I can ssh in.

Not a big deal since everything works as expected if I keep the setting disabled but just wanted to report it here since it seems to be unintended behaviour.

antheas, 2024-11-22:
Yeah well we kind of cant roll it out with this bug can we

I made a potential fix

If you clone the following on a different linux computer:
https://github.com/antheas/gamescope/tree/gamescope-ba

And run ./sync.sh you will be able to test the fix. Otherwise, it should be in the next testing build

antheas, 2024-11-23:
Update is up

SimpleHeuristics, 2024-11-24:
> Update is up on testing

Seems to have fixed it!

What is the difference though between this setting being on or off? The Win Mini didn't have issues with suspend and resume before and it seemed to be fully asleep previously as well with almost no battery drain.

antheas, 2024-11-24:
The purpose is to turn off the screen sooner, around .5s or so, along with the controller, RGB of the device, and power light of the device.

Otherwise, the kernel turns off the screen and you get a frozen framebuffer for around 1 second.

The result looks nicer primarily.

It also allows initiating hibernation without having the screen be on

closed 2024-11-28 by antheas