hhd-dev/hhd #116: Installation on Linux Mint 21.3

shade-belisar, 2024-11-20:
I'm trying to install hhd on a GPD Win MAX 2 (2024) running Linux Mint 21.3. However, the script never finishes. Instead, it goes through all versions starting from 3.6.0 down, breaking of each time with the message: "inconsistent name: filename has 'hhd', but metadata has 'unknown'".
After installing all dependencies I ran this command:
`curl -L https://github.com/hhd-dev/hhd/raw/master/install.sh | bash`
And got this output (cut for brevity, but it just goes on along the same lines):
```
Collecting hhd
  Using cached hhd-3.6.0.tar.gz (264 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package hhd produced metadata for project name unknown. Fix your #egg=hhd fragments.
Discarding https://files.pythonhosted.org/packages/bb/33/2b65ead17297d2761c8ab7b20b259d258cd53021712ebf7c4a67c77fd4ee/hhd-3.6.0.tar.gz#sha256=0464775e100ead25604f7e9594c8cb37051cf6af953e9e3036ab9a6864e0c405 (from https://pypi.org/simple/hhd/) (requires-python:>=3.10): Requested unknown from https://files.pythonhosted.org/packages/bb/33/2b65ead17297d2761c8ab7b20b259d258cd53021712ebf7c4a67c77fd4ee/hhd-3.6.0.tar.gz#sha256=0464775e100ead25604f7e9594c8cb37051cf6af953e9e3036ab9a6864e0c405 has inconsistent name: filename has 'hhd', but metadata has 'unknown'
  Using cached hhd-3.5.12.tar.gz (262 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package hhd produced metadata for project name unknown. Fix your #egg=hhd fragments.
Discarding https://files.pythonhosted.org/packages/75/42/7ebcf1746c926661b0d542a145bacc0acc4b3ed321f527bb3abc8ed386fd/hhd-3.5.12.tar.gz#sha256=79a1de64fa55347e49dfaa3c5de4b2e0dd8a70ff6001505ee9ff4aca5ec7c1ad (from https://pypi.org/simple/hhd/) (requires-python:>=3.10): Requested unknown from https://files.pythonhosted.org/packages/75/42/7ebcf1746c926661b0d542a145bacc0acc4b3ed321f527bb3abc8ed386fd/hhd-3.5.12.tar.gz#sha256=79a1de64fa55347e49dfaa3c5de4b2e0dd8a70ff6001505ee9ff4aca5ec7c1ad has inconsistent name: filename has 'hhd', but metadata has 'unknown'
  Using cached hhd-3.5.11.tar.gz (262 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
  WARNING: Generating metadata for package hhd produced metadata for project name unknown. Fix your #egg=hhd fragments.
Discarding https://files.pythonhosted.org/packages/5c/94/3dec77b0fb0604af219167c9a44b99bdce2a624991aa6c65851df1c35eea/hhd-3.5.11.tar.gz#sha256=be2c17e3046d54af82f87b2d589c0b7ebb548d3d4218af4f39b5e27de9509de9 (from https://pypi.org/simple/hhd/) (requires-python:>=3.10): Requested unknown from https://files.pythonhosted.org/packages/5c/94/3dec77b0fb0604af219167c9a44b99bdce2a624991aa6c65851df1c35eea/hhd-3.5.11.tar.gz#sha256=be2c17e3046d54af82f87b2d589c0b7ebb548d3d4218af4f39b5e27de9509de9 has inconsistent name: filename has 'hhd', but metadata has 'unknown'
...
```
Is this a bug, or am I using the script wrong?

antheas, 2024-11-20:
You are using python 3.8. Make sure you are using python 3.10, ie the mint you are using is based on ubuntu 24. Search for the PopOS issue for more details

> Linux Mint 21.3 features a Linux kernel 5.15 and an Ubuntu 22.04 package base.

closed 2024-11-20 by antheas

shade-belisar, 2024-11-20:
I'm afraid that is not correct. When I enter `python3 --version` I get `Python 3.10.12` back. I also read through the Pop!OS issue. Did you mean I'm running python 3.10 but 3.12 is required?

antheas, 2024-11-20:
I haven't tested hhd on python 3.10 for a while unfortunately. I've only done 3.11+ recently. But could have sworn Ubuntu 24 works and Ubuntu 22 does not because of the python version 

In any case, you can use the deadsnakes ppa and run the commands by hand to use python 3.12 which is the current ver hhd is tested wkth