hhd-dev/hhd #118: Open Steam if its not running in desktop mode

fennectech, 2024-12-02:
Can we assign the left steam button to open steam if it has not already started. Useful in desktop mode if steam crashes or hasn’t started with the system.  This would solve the user needing to shift grips to touch the screen if steam isnt running and providing desktop controls in desktop mode.

antheas, 2024-12-26:
The problem here is that processes need a parent and hhd should not be steam's parent. It should be owned by the current session. I.e., you need to make it autostart.

Also, it never crashes in desktop mode

But I am aware the shortcuts are quite useless in desktop mode currently

fennectech, 2024-12-31:
Ive had times when steam failed to start on login for what ever reason.  I have to go hunt for it in the menus or desktop then.   It should be possible to start and then disown steam. So it becomes owned by the session.

antheas, 2025-01-09:
hhd is not part of the session. The problem is that steam needs to be under the session leader process, not disowned

antheas, 2025-06-30:
Too difficult, you should set it so it autostarts.

closed 2025-06-30 by antheas