hhd-dev/hhd #119: [Legion Go] LED color related to battery life

SeeDborg, 2024-12-03:
Hello and thanks for this wonderful software.

I was wondering if an option could be added : the LED color from red to green, related to battery life. It would be so great to see the state of the battery without having to turn the device on !
The option would even be possible when the console is an (2 different options ??)

Let me know if that would be possible :)

SeeDborg, 2024-12-20:
Good idea, bad idea ? Doable ? :)

antheas, 2024-12-20:
Unfortunately, the power button LED is hardcoded in the firmware

SeeDborg, 2024-12-27:
Sad, but I understand :(
Would it be possible to light it up when the console is charging ? (Like on Deck)

antheas, 2024-12-27:
Doesn't the legion go have a charge light indicator? I don't have it with me but I'm pretty sure it did

jerem62, 2024-12-30:
Nice idea, for the rog ally, power light are only white, may be simply change the stick rgb color instead of the power light ? 

If i'm not wrong, legion go has also these light, so it should be appliable also

SeeDborg, 2025-08-22:
That is not exactly what we want ^^